NVlabs/Deep_Object_Pose #95: Getting error on train.py

okadahiroyuki, 2020-01-13:
Hi,
I want to try training with other objects.
The following error messages are displayed on terminal.
Any help would be highly appreciated.

Ubuntu 18.04.3 LTS (docker image: nvidia-dope:kinetic-v1)
memory: 62.5 GiB
Intel® Core™ i9-9880H CPU @ 2.30GHz × 16
GeForce GTX 1650/PCIe/SSE2

$ ./run_dope_docker.sh
root@X1:~/catkin_ws/src/dope/scripts# python train.py --data /home/abdu/fat --object 037_scissors_16k --outf 037_scissors_16k --gpuids 0

start: 07:05:15.913536                                                        
load data
training data: 7688 batches
load models
Training network pretrained on imagenet.

Traceback (most recent call last):
File "train.py", line 1393, in <module>                                                      
    _runnetwork(epoch,trainingdata)                                                            
  File "train.py", line 1335, in _runnetwork                                                   
    output_belief, output_affinities = net(data)                                               
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/modules/module.py", line 491, in __call
__                                                                                             
    result = self.forward(*input, **kwargs)                                                    
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/parallel/data_parallel.py", line 112, i
n forward                                                                                      
    return self.module(*inputs[0], **kwargs[0])                                                
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/modules/module.py", line 491, in __call__
    result = self.forward(*input, **kwargs)
  File "train.py", line 154, in forward
    out1 = self.vgg(x)
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/modules/module.py", line 491, in __call__
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/modules/container.py", line 91, in forward
    input = module(input)
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/modules/module.py", line 491, in __call__
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/torch/nn/modules/conv.py", line 301, in forward
    self.padding, self.dilation, self.groups)
RuntimeError: CUDNN_STATUS_EXECUTION_FAILED

michalvavrecka, 2020-01-14:
This is probably wrong version of Cudnn, works with CUDA 8.0 and Cudnn 7.1.4 . You also need at least 2 GPU to train as it takes 17 GB GPU during training :(

avinashsen707, 2020-01-14:
> This is probably wrong version of Cudnn, works with CUDA 8.0 and Cudnn 7.1.4 . You also need at least 2 GPU to train as it takes 17 GB GPU during training :(

dos this support in  this following version

```
` nvidia-smi
Wed Jan 15 01:28:56 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 418.39       Driver Version: 418.39       CUDA Version: 10.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  Quadro P4000        Off  | 00000000:B3:00.0  On |                  N/A |
| 47%   39C    P8     8W / 105W |    404MiB /  8111MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1814      G   /usr/lib/xorg/Xorg                           280MiB |
|    0      2482      G   compiz                                        63MiB |
|    0      2783      G   ...quest-channel-token=3259903450700816574    58MiB |
+-----------------------------------------------------------------------------+
WARNING: infoROM is corrupted at gpu 0000:B3:00.0
```
`

okadahiroyuki, 2020-01-15:
Wed Jan 15 00:05:20 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.44       Driver Version: 440.44       CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 1650    Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   40C    P8     2W /  N/A |    336MiB /  3911MiB |      2%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
+-----------------------------------------------------------------------------+

>You also need at least 2 GPU to train as it takes 17 GB GPU during training :(
!!!

TontonTremblay, 2020-01-17:
You might be able to train with a batchsize of 2, but it is going to take a very long time.

okadahiroyuki, 2020-01-17:
> You might be able to train with a batchsize of 2, but it is going to take a very long time.
Tkanks!
I`ll try

Bochicchio3, 2020-03-16:
Hi,
I am trying to train on a 1660ti (6gb) but i get the "RuntimeError: CUDNN_STATUS_EXECUTION_FAILED" error. I am using the docker image and I have tried to decrease both batch size and epochs without success. I don't understand if I am running out of memory or if there is some other error. 
 Do you have any suggestion on what I can try next? How could i test if it is the memory or some other problem?
Thank you!

TontonTremblay, 2020-03-17:
This does not seem to be an error for out of memory. I never played around with a 1660ti, thus my experience is limited on it.

avinashsen707, 2020-03-18:
I think you can try in Google Colab.

[https://colab.research.google.com/](url)