NVlabs/Deep_Object_Pose #96: perceived distance error

huckl3b3rry87, 2020-01-17:
I am not getting anywhere near the accuracy that is reported in the paper when it comes to the pose. 

For instance consider the case where my camera is roughly 80cm away from the soup can in the Z direction
[image]

while the pose estimate looks good on the image, the output is
```
  frame_id: "kinect2_1_ir_optical_frame"
pose: 
  position: 
    x: -1.50516920918
    y: -0.582944770056
    z: 2.19036338323
  orientation: 
    x: 0.499659802676
    y: -0.608086512688
    z: -0.15483560795
    w: 0.59715727341
```
so, the estimate for z is 2.19, which is way off. 

I have used the default settings when possible, but some differences to my current setup are:
* Soup can model trained to 16 epochs.
* Kinect camera using ``/kinect2_1/sd/image_color_rect`` 
  --> which I re-calibrated, with no improvement

Do I really need to train to 60 epoch? Or am I making an error somewhere else? 

I assumed that these mesh scales should be given as https://github.com/NVlabs/Deep_Object_Pose/blob/master/config/config_pose.yaml#L71 since I did not change any mesh files or training setup.

TontonTremblay, 2020-01-17:
Did you try to train new weights for the soup can? The pose displayed is also some what crooked, so even the orientation is not correct. I think DOPE is not using the right intrinsic for the camera, it looks like you are using focal values that are somewhat larger than in real life. Can you look at the camera_info of your kinect to make sure they are correct? I dont have access to a kinect 2 easily so I cannot test the camera. Can I suggest you test everything with a different camera just in case?

huckl3b3rry87, 2020-01-17:
1. Did you try to train new weights for the soup can? -> yes to 16 epochs.

I think that I figured out the issue, I changed the [downscale_height](https://github.com/NVlabs/Deep_Object_Pose/blob/master/config/config_pose.yaml#L5) parameter from 500 to 300 in an attempt to make it run faster. Now it looks like this 
[image]

and the pose is 
```
pose: 
  position: 
    x: -0.0110320733664
    y: 0.0753695222389
    z: 0.863267260191
  orientation: 
    x: 0.00707828556264
    y: 0.946789174063
    z: 0.117598448411
    w: 0.299517549872
```
 which, as far as I can tell from a few quick measurements, is accurate.

Based off of this it seems that the downscale_height parameter should not be adjusted. Is this correct?

mintar, 2020-01-18:
> Based off of this it seems that the downscale_height parameter should not be adjusted. Is this correct?

No. The correct statement is: If the object appears too small in the downscaled image, you will get bad results. You can see this effect in your first image: The object is just a pixelated mess, so I'm amazed that DOPE recognizes it at all - it's not surprising that there is a large error in the estimated orientation (garbage in, garbage out).

How large your object appears in the downscaled image is a combination of several factors:

* distance from camera to object
* camera field of view (= how much your camera zooms in)
* the `downscale_height` parameter

In general, the object should be in the same effective size range (in pixels) in the downscaled image as it had in the dataset that DOPE was trained on. If your object appears too small or too large (!), the results will get worse.

xdw-123, 2020-04-14:
How do you output the position and attitude information of this model？

TontonTremblay, 2020-04-14:
Do you mean the pose? like translation and rotation?

avinashsen707, 2020-04-16:
> How do you output the position and attitude information of this model？

In rostopic lists and ros node lists , you can see lot of nodes and topic.

Eirher you can "echo rostopic info __________" followed by the required one can find solutions.