NVlabs/Deep_Object_Pose #97: FileNotFoundError: [Errno 2] No such file or directory: './cube/' #how to train the cyustom dataset?

avinashsen707, 2020-01-31:
```
binpick@ncrai-Precision-7820-Tower:~/catkin_ws/src/dope/scripts$ python3 train.py --data ./cube/ --outf cube_1214  --gpuids 0 1 --epochs 120 --loginterval 1 --batchsize 32
start: 00:11:52.635052
load data
Traceback (most recent call last):
  File "train.py", line 1255, in <module>
    transforms.Scale(opt.imagesize//8),
  File "train.py", line 416, in __init__
    self.imgs = load_data(root)
  File "train.py", line 411, in load_data
    for name in os.listdir(str(path)):
FileNotFoundError: [Errno 2] No such file or directory: './cube/'

```

[image]

TontonTremblay, 2020-01-31:
Can you give it the absolute path without the `.`? Some thing like `/home/jtremblay/data/cude/`

avinashsen707, 2020-02-01:
> Can you give it the absolute path without the `.`? Some thing like `/home/jtremblay/data/cude/`

```
/media/binpick/DATA/binpick_extended_user/Training$ python3 train.py --data ./fat/single/011_banana_16k/kitchen_0/ --outf ban
ana  --gpuids 0 1 --epochs 120 --loginterval 1 --batchsize 32
start: 11:08:10.511437
load data
training data: 7 batches
load models
Training network pretrained on imagenet.
Traceback (most recent call last):
  File "train.py", line 1306, in <module>
    net = torch.nn.DataParallel(net,device_ids=opt.gpuids).cuda()
  File "/home/binpick/.local/lib/python3.5/site-packages/torch/nn/parallel/data_parallel.py", line 102, in __init__
    _check_balance(self.device_ids)
  File "/home/binpick/.local/lib/python3.5/site-packages/torch/nn/parallel/data_parallel.py", line 17, in _check_balance
    dev_props = [torch.cuda.get_device_properties(i) for i in device_ids]
  File "/home/binpick/.local/lib/python3.5/site-packages/torch/nn/parallel/data_parallel.py", line 17, in <listcomp>
    dev_props = [torch.cuda.get_device_properties(i) for i in device_ids]
  File "/home/binpick/.local/lib/python3.5/site-packages/torch/cuda/__init__.py", line 292, in get_device_properties
    raise AssertionError("Invalid device id")
AssertionError: Invalid device id
```

Thankyou sir, while i look in the path and corrected ; it ended up in the following error!

TontonTremblay, 2020-02-06:
the batchsize is too large. Try `--batchsize` 1.

avinashsen707, 2020-02-06:
> the batchsize is too large. Try `--batchsize` 1.

thankyou sir, it worked. 
I am training now with FAT dataset for each object. But the confusion is that there are almost 20 background for each object; **so i need to do train in each folder to get weights. So how can i select the final  weight to be placed in DOPE code**; _since i am getting alot of weights as per my given epoch in each folder._

and could you help in giving the optimum epoch and log interval does i need to give to get good results, i am using Nvidia P4000 quadro gpu with 8GB in dell precision7820 workstation.

Thanking you in advance for the patient reply for beginers like me.

TontonTremblay, 2020-02-06:
With 8gb you should be able to run a batchsize of 8, the idea here is to increase it until you have fill your gpu memory. You can check with nvidia-smi while it is running to see how much you are using. 

You can give `/fat/` to the training script with the `--objectofinterest banana` it will only use the banana information. 

I hope this helps.

avinashsen707, 2020-02-07:
as per your suggestion i made it like this

```
binpick@ncrai-Precision-7820-Tower:/media/binpick/DATA/binpick_userfiles/Training$ python3 train.py --data ./pvc_tee_800/ --outf tee_1214  --gpuids 0 --epochs 10 --loginterval 1 --batchsize 8
start: 09:55:13.370283
load data
training data: 100 batches
load models
Training network pretrained on imagenet.
Traceback (most recent call last):
  File "train.py", line 1392, in <module>
    _runnetwork(epoch,trainingdata)
  File "train.py", line 1330, in _runnetwork
    for batch_idx, targets in enumerate(loader):
  File "/home/binpick/.local/lib/python3.5/site-packages/torch/utils/data/dataloader.py", line 286, in __next__
    return self._process_next_batch(batch)
  File "/home/binpick/.local/lib/python3.5/site-packages/torch/utils/data/dataloader.py", line 307, in _process_next_batch
    raise batch.exc_type(batch.exc_msg)
IndexError: Traceback (most recent call last):
  File "/home/binpick/.local/lib/python3.5/site-packages/torch/utils/data/dataloader.py", line 57, in _worker_loop
    samples = collate_fn([dataset[i] for i in batch_indices])
  File "/home/binpick/.local/lib/python3.5/site-packages/torch/utils/data/dataloader.py", line 57, in <listcomp>
    samples = collate_fn([dataset[i] for i in batch_indices])
  File "train.py", line 491, in __getitem__
    cuboid = np.array(data['exported_objects'][0]['cuboid_dimensions'])
**IndexError: list index out of range**

```

avinashsen707, 2020-03-04:
Problem solved ; the json files data were missing inside it to load for the script!

closed 2020-03-04 by avinashsen707

tuhinmallick, 2021-06-18:
Traceback (most recent call last):
  File "train.py", line 1312, in <module>
    net = torch.nn.DataParallel(net,device_ids=opt.gpuids).cuda()
  File "C:\Users\intraflyQuadro\anaconda3\lib\site-packages\torch\nn\parallel\data_parallel.py", line 142, in __init__
    _check_balance(self.device_ids)
  File "C:\Users\intraflyQuadro\anaconda3\lib\site-packages\torch\nn\parallel\data_parallel.py", line 23, in _check_balance
    dev_props = _get_devices_properties(device_ids)
  File "C:\Users\intraflyQuadro\anaconda3\lib\site-packages\torch\_utils.py", line 459, in _get_devices_properties
    return [_get_device_attr(lambda m: m.get_device_properties(i)) for i in device_ids]
  File "C:\Users\intraflyQuadro\anaconda3\lib\site-packages\torch\_utils.py", line 459, in <listcomp>
    return [_get_device_attr(lambda m: m.get_device_properties(i)) for i in device_ids]
  File "C:\Users\intraflyQuadro\anaconda3\lib\site-packages\torch\_utils.py", line 442, in _get_device_attr
    return get_member(torch.cuda)
  File "C:\Users\intraflyQuadro\anaconda3\lib\site-packages\torch\_utils.py", line 459, in <lambda>
    return [_get_device_attr(lambda m: m.get_device_properties(i)) for i in device_ids]
  File "C:\Users\intraflyQuadro\anaconda3\lib\site-packages\torch\cuda\__init__.py", line 309, in get_device_properties
    raise AssertionError("Invalid device id")
AssertionError: Invalid device id

How did you get rid of this error ?

TontonTremblay, 2021-06-18:
how many GPUs do you have? What is the value of `opt.gpuids`?

tuhinmallick, 2021-06-24:
I have only one GPU (  Quadro RTX 6000) and I got the device id fixed by specifying the gpuid as only 0. thanks for the help.
 But now I am getting the following error 

Training network pretrained on imagenet.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\intraflyQuadro\anaconda3\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Users\intraflyQuadro\anaconda3\lib\multiprocessing\spawn.py", line 125, in _main
    prepare(preparation_data)
  File "C:\Users\intraflyQuadro\anaconda3\lib\multiprocessing\spawn.py", line 236, in prepare
    _fixup_main_from_path(data['init_main_from_path'])
  File "C:\Users\intraflyQuadro\anaconda3\lib\multiprocessing\spawn.py", line 287, in _fixup_main_from_path
    main_content = runpy.run_path(main_path,
  File "C:\Users\intraflyQuadro\anaconda3\lib\runpy.py", line 265, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "C:\Users\intraflyQuadro\anaconda3\lib\runpy.py", line 97, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "C:\Users\intraflyQuadro\anaconda3\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\intraflyQuadro\Desktop\Deep_Object_Pose\scripts\train.py", line 1398, in <module>
    _runnetwork(epoch,trainingdata)
  File "C:\Users\intraflyQuadro\Desktop\Deep_Object_Pose\scripts\train.py", line 1336, in _runnetwork
    for batch_idx, targets in enumerate(loader):
  File "C:\Users\intraflyQuadro\anaconda3\lib\site-packages\torch\utils\data\dataloader.py", line 355, in __iter__
    return self._get_iterator()
  File "C:\Users\intraflyQuadro\anaconda3\lib\site-packages\torch\utils\data\dataloader.py", line 301, in _get_iterator
    return _MultiProcessingDataLoaderIter(self)
  File "C:\Users\intraflyQuadro\anaconda3\lib\site-packages\torch\utils\data\dataloader.py", line 914, in __init__
    w.start()
  File "C:\Users\intraflyQuadro\anaconda3\lib\multiprocessing\process.py", line 121, in start
    self._popen = self._Popen(self)
  File "C:\Users\intraflyQuadro\anaconda3\lib\multiprocessing\context.py", line 224, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "C:\Users\intraflyQuadro\anaconda3\lib\multiprocessing\context.py", line 327, in _Popen
    return Popen(process_obj)
  File "C:\Users\intraflyQuadro\anaconda3\lib\multiprocessing\popen_spawn_win32.py", line 45, in __init__
    prep_data = spawn.get_preparation_data(process_obj._name)
  File "C:\Users\intraflyQuadro\anaconda3\lib\multiprocessing\spawn.py", line 154, in get_preparation_data
    _check_not_importing_main()
  File "C:\Users\intraflyQuadro\anaconda3\lib\multiprocessing\spawn.py", line 134, in _check_not_importing_main
    raise RuntimeError('''
RuntimeError:
        An attempt has been made to start a new process before the
        current process has finished its bootstrapping phase.

        This probably means that you are not using fork to start your
        child processes and you have forgotten to use the proper idiom
        in the main module:

            if __name__ == '__main__':
                freeze_support()
                ...

        The "freeze_support()" line can be omitted if the program
        is not going to be frozen to produce an executable.

Can you please help me out ?

TontonTremblay, 2021-06-24:
I never seen this error, and I never tested a quadro with this code. Can you try to change the number of workers on the dataloader to 1?

tuhinmallick, 2021-07-30:
The problem did get sorted out by having the number of workers on the dataloader to 0. 

But a new error cropped up :- 

[image]

TontonTremblay, 2021-07-30:
Are you using a dataset from.nvisii?

tuhinmallick, 2021-07-30:
No, I am using NDDS