NVlabs/Deep_Object_Pose #98: NameError: global name 'Quaternion' is not defined

avinashsen707, 2020-02-06:
[image]

```
binpick@ncrai-Precision-7820-Tower:~/catkin_ws$ roslaunch dope dope.launch
... logging to /home/binpick/.ros/log/db3c021a-49be-11ea-a8a9-6c2b59ca96b5/roslaunch-ncrai-Precision-7820-Tower-9847.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ncrai-Precision-7820-Tower:40553/

SUMMARY
========

CLEAR PARAMETERS
 * /dope/

PARAMETERS
 * /dope/class_ids/bleach: 7
 * /dope/class_ids/bottle: 5
 * /dope/class_ids/cracker: 1
 * /dope/class_ids/gelatin: 2
 * /dope/class_ids/meat: 3
 * /dope/class_ids/mustard: 4
 * /dope/class_ids/sugar: 6
 * /dope/dimensions/bleach: [10.2677307128906...
 * /dope/dimensions/bottle: [6.76593780517578...
 * /dope/dimensions/cracker: [16.4036006927490...
 * /dope/dimensions/gelatin: [8.91829967498779...
 * /dope/dimensions/meat: [10.1646738052368...
 * /dope/dimensions/mustard: [9.60241508483886...
 * /dope/dimensions/sugar: [9.26773071289062...
 * /dope/downscale_height: 500
 * /dope/draw_colors/bleach: [232, 222, 12]
 * /dope/draw_colors/bottle: [255, 101, 0]
 * /dope/draw_colors/cracker: [13, 255, 128]
 * /dope/draw_colors/gelatin: [255, 255, 255]
 * /dope/draw_colors/meat: [0, 104, 255]
 * /dope/draw_colors/mustard: [217, 12, 232]
 * /dope/draw_colors/sugar: [232, 222, 12]
 * /dope/input_is_rectified: True
 * /dope/mesh_scales/bleach: 0.01
 * /dope/mesh_scales/bottle: 0.01
 * /dope/mesh_scales/cracker: 0.01
 * /dope/mesh_scales/gelatin: 0.01
 * /dope/mesh_scales/meat: 0.01
 * /dope/mesh_scales/mustard: 0.01
 * /dope/mesh_scales/sugar: 0.01
 * /dope/sigma: 3
 * /dope/thresh_angle: 0.5
 * /dope/thresh_map: 0.01
 * /dope/thresh_points: 0.1
 * /dope/topic_camera: /camera/color/ima...
 * /dope/topic_camera_info: /camera/color/cam...
 * /dope/topic_publishing: dope
 * /dope/weights/bottle: package://dope/we...
 * /dope/weights/cracker: package://dope/we...
 * /dope/weights/gelatin: package://dope/we...
 * /dope/weights/meat: package://dope/we...
 * /dope/weights/mustard: package://dope/we...
 * /dope/weights/sugar: package://dope/we...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    dope (dope/dope)

ROS_MASTER_URI=http://localhost:11311

process[dope-1]: started with pid [9867]
Loading DOPE model '/home/binpick/catkin_ws/src/dope/weights/mustard_60.pth'...
    Model loaded in 6.73668289185 seconds.
Loading DOPE model '/home/binpick/catkin_ws/src/dope/weights/cracker_60.pth'...
    Model loaded in 3.24266910553 seconds.
Loading DOPE model '/home/binpick/catkin_ws/src/dope/weights/sugar_60.pth'...
    Model loaded in 3.12896490097 seconds.
Loading DOPE model '/home/binpick/catkin_ws/src/dope/weights/gelatin_60.pth'...
    Model loaded in 3.11447501183 seconds.
Loading DOPE model '/home/binpick/catkin_ws/src/dope/weights/soup_60.pth'...
    Model loaded in 3.11263012886 seconds.
Loading DOPE model '/home/binpick/catkin_ws/src/dope/weights/meat_20.pth'...
    Model loaded in 3.09068608284 seconds.
Running DOPE...  (Listening to camera topic: '/camera/color/image_rect_color')
Ctrl-C to stop
[ERROR] [1581095733.719928]: bad callback: <bound method Subscriber.callback of <message_filters.Subscriber object at 0x7f7ed76ec310>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 75, in callback
    self.signalMessage(msg)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 57, in signalMessage
    cb(*(msg + args))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 224, in add
    self.signalMessage(*msgs)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 57, in signalMessage
    cb(*(msg + args))
  File "/home/binpick/catkin_ws/src/dope/nodes/dope", line 248, in image_callback
    self.config_detect
  File "/home/binpick/catkin_ws/src/dope/nodes/detector.py", line 257, in detect_object_in_image
    detected_objects = ObjectDetector.find_object_poses(vertex2, aff, pnp_solver, config)
  File "/home/binpick/catkin_ws/src/dope/nodes/detector.py", line 274, in find_object_poses
    location, quaternion, projected_points = pnp_solver.solve_pnp(points)
  File "/home/binpick/catkin_ws/src/dope/nodes/cuboid_pnp_solver.py", line 98, in solve_pnp
    quaternion = self.convert_rvec_to_quaternion(rvec)
  File "/home/binpick/catkin_ws/src/dope/nodes/cuboid_pnp_solver.py", line 122, in convert_rvec_to_quaternion
    return Quaternion.from_axis_rotation(raxis, theta)
NameError: global name 'Quaternion' is not defined

```

Getting error on detecting the object in camera frame!!
at that time in next tab shows as..

```
binpick@ncrai-Precision-7820-Tower:~/catkin_ws$ rosrun rviz rviz[ INFO] [1581095616.327188562]: rviz version 1.12.17
[ INFO] [1581095616.327353590]: compiled against Qt version 5.5.1
[ INFO] [1581095616.327440963]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1581095618.595676324]: Stereo is NOT SUPPORTED
[ INFO] [1581095618.596096659]: OpenGl version: 4.5 (GLSL 4.5).
[ WARN] [1581095708.123672254]: MessageFilter [target=camera_color_frame ]: Discarding message from [unknown] due to empty frame_id.  This message will only print once.
[ WARN] [1581095721.967220519]: MessageFilter [target=camera_color_frame ]: Discarding message from [unknown_publisher] due to empty frame_id.  This message will only print once.

```

@mintar @foertter @tmbdev @vinodgro @v4hn  Can you please help me!

TontonTremblay, 2020-02-06:
Do you have pyquaternion installed?

avinashsen707, 2020-02-06:
> Do you have pyquaternion installed?

https://pypi.org/project/pyquaternion/
 i installed using pip3 and pip2.

TontonTremblay, 2020-02-10:
Which version of pyquaternion do you have installed?

avinashsen707, 2020-02-11:
`binpick@ncrai-Precision-7820-Tower:~$ pip3 install pyquaternion`

Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pyquaternion in ./.local/lib/python3.5/site-packages (0.9.5)
Requirement already satisfied: numpy in ./.local/lib/python3.5/site-packages (from pyquaternion) (1.14.2)

avinashsen707, 2020-03-04:
problem solved , was due to the mismatch pf pytorch and torch vision versions installed.

closed 2020-03-04 by avinashsen707