NVlabs/Deep_Object_Pose #99: using DOPE on HSR robot

ROSNovice, 2020-02-09:
Hi. in which parameters or file, if i want to use DOPE in another type of camera or robot?. should i make camera calibration before?  i tried only change config file for kinect, it was successful, but when i use dope in HSR robot, it can not show the cuboid.

TontonTremblay, 2020-02-10:
If you use a camera node in ros, you should be correct, just change the topic in the config.yaml file to listen to your camera image. The script should look for the calibration file if there is one, if not you should calibrate. We have quite an extensive tutorial on how to do this in the documentation.

ROSNovice, 2020-02-10:
thanks for your reply, 
so in my understanding from your explanation, if I do not make a camera on my robot camera, it will not show the cuboid in my HSR robot camera, isn't it? in my case, the dope node keeps listening and wait for the robot camera topic. For your information, HSR is based on ROS kinetic

TontonTremblay, 2020-02-10:
I am not sure anymore of what you are asking. Could you describe your set up a little bit further please?

ROSNovice, 2020-02-10:
OK, I will explain a little bit more detail. i am sorry i am not good in english
First, I only change the setting of the camera topic and camera info in the config_pose.yaml file without change camera_info.yaml file.
###################################
topic_camera: "/hsrb/hand_camera/image_raw"
topic_camera_info: "/hsrb/hand_camera/camera_info"
#######################################

then, i run roslaunch for HSR robot hand camera, then i run roslaunch dope dope.launch [config:=/path/to/my_config.yaml]  # Config file is optional; default is `config_pose.yaml'

and then i opened rviz to show the visualization of the detected object. but when i change the image topic to the dope/rgb_points, the screen becomes "no image" . Moreover, when i look at the terminal of roslaunch dope, it still keeps like this : 

Listening to camera topic: (/hsrb/hand_camera/image_raw) Ctrl-C to stop

i do not have any idea about this problem. if this problem due to need calibration in advance, i will try to do that.

TontonTremblay, 2020-02-11:
Ok good, I do not think you are having problems with the camera info. There is a problem where DOPE does not get any images. Can you find the image_callback function and add a print statement in there to see if it sees an image.

mintar, 2020-02-11:
>  but when i change the image topic to the dope/rgb_points, the screen becomes "no image"

That sounds like the problem I've solved here: https://github.com/NVlabs/Deep_Object_Pose/issues/70#issuecomment-525652741

In short: You cannot use the topic `dope/rgb_points` with the `Camera` RViz display, because it doesn't have a corresponding `dope/camera_info` topic.

* Either use the topic `/hsrb/hand_camera/image_raw` with the `Camera` display
* Or use the `dope/rgb_points` topic with the `Image` display

@TontonTremblay : We should probably really publish a `dope/camera_info` topic. Many people are getting tripped up by this.

ROSNovice, 2020-02-11:
> That sounds like the problem I've solved here: #70 (comment)

Hi. Thank you for your advice. i have tried using image_raw. the frame rate
is very slow but the rgb points is not appeared yet. When i used dope rgb
points, it is indicated no image. Maybe i have to try using rectified. i
will try soon and inform to you all.

ROSNovice, 2020-02-12:
hi @mintar  actually your method is work only on my kinect and my webcam. but in Toyota HSR camera robot it is still trouble and indicates no image. when i try to use rostopic echo for crackers points it also nothing happen. and this is the result 
[image]

ROSNovice, 2020-02-12:
an this one the result using webcamera
[image]
camera