NVlabs/Deep_Object_Pose #101: Testing using ADD Metric

IbrahimMCode, 2020-03-31:
Hello, 

Is there any implementation for testing the trained model using the ADD Metric as written in the paper.
The test data argument that can be added to the training calculates the loss using Affinities and Belief maps.

Thank you

TontonTremblay, 2020-04-01:
Hello, we used the add code provided by PoseCNN and the ycb videos toolkit. You can look into it. 

Also for the axes display (the other question I have an email for, but I cannot find on github), we have a webcam version (internal for now) that uses opengl to display the 3d model from which we capture the projected axes. You could reproduce this in ROS byt re-projecting local axes using something like `projected_points, _ = cv2.projectPoints(cuboid3d_points, rvec, tvec, self._camera_intrinsic_matrix, self._dist_coeffs)` Where you need to change `cuboid3d_points` to the axis, which should be fairly simple. I hope this helps.

mintar, 2020-04-01:
You can't find the question on Github because he deleted the comment. Anyway, you can already visualize the axes with the current version of DOPE: simply visualize the pose topics in RViz.

IbrahimMCode, 2020-04-01:
Hello, 
For the axes display, I was already able to draw them.
But for the Testing, do you have a reference to where I can test the trained Dope model?
I also think that the formats are somehow different between PoseCNN and Dope, so should there be a conversion or something similar?

Thank you in advance

TontonTremblay, 2020-04-01:
Yu ended up doing that part for the paper and we did not include that code with this release. I am currently working on a version that would work with ndds format annotation. But I dont have a timeline for it right now.

IbrahimMCode, 2020-04-02:
I referred to another issue to calculate the ADD error cuboid but I was stuck into guessing the projected_cuboid unit used in the FAT Dataset labels(around 400 or 500). Can you please tell me which unit is used there?
Thank you

TontonTremblay, 2020-04-02:
We used m in the paper for calculating the ADD. But as long as you have the units the same everywhere, you should be able to change them.

IbrahimMCode, 2020-04-02:
Sorry for bothering you, I'm still confused, do you mean that in the FAT Dataset the projected cuboids are in meters? Because as I've seen in the _object_settings.json, for example, the mug cuboid dimensions are in centimeters.
I'm trying to calculate the cuboid error between the real pose and the predicted ones, I have a result but I still can't guess the unit.

Thank you

TontonTremblay, 2020-04-02:
Everything is in cm for FAT/NDDS/DOPE

IbrahimMCode, 2020-05-18:
Hello,

In the paper's experiments the maximum error value in the ADD Metric curve is 10cm and the threshold is 2cm(level of accuracy necessary for grasping using our robotic manipulator (2 cm)).
I Just have one question, if i have an object of 100x60x50 for example should i change the maximum value used in the paper(compared to the FAT Dataset objects, my object is way bigger)?
Also if this object is type of symmetric(with some indications to the front), when calculating the add metric between the ground truth pose and the wrongly rotated predicted pose, should i classify predictions having an add result larger than the width of this object as a bad rotation?

TontonTremblay, 2020-05-18:
As for the 10cm, this was used only to compared the results with prior art in the domain. If you want you could 100cm. You should look at the ADD-S for reporting results on symmetrical objects.

BigRays, 2021-01-05:
> As for the 10cm, this was used only to compared the results with prior art in the domain. If you want you could 100cm. You should look at the ADD-S for reporting results on symmetrical objects.

Can you provide the code for testing with ADD？

DOPE-liyulong, 2022-03-22:
> 你好， 对于轴显示，我已经能够绘制它们。 但是对于测试，你有没有参考我可以在哪里测试经过训练的 Dope 模型？ 我还认为 PoseCNN 和 Dope 之间的格式有些不同，所以应该进行转换或类似的东西吗？
> 
> 先感谢您

Hello, first of all, thank you for your work. How do you test the trained model?