NVlabs/Deep_Object_Pose #102: A few important questions about custom dataset training

yyc9268, 2020-04-02:
Hello TontonTremblay,
It is pretty exciting to utilize decent opensource.
We are trying to use the DOPE for our own target object.
I have analyzed the code so far.
And finally stuck in following a few questions.

1. The main advantage of this simple and strong network is from NDDS based augmentation.
However, the NDDS repository is not accessible since 'git lfs data limit reached'.
I guess the problem can only be solved by repository manager.

2. There are a few difference in input data settings between training script and test script.
In test script, the input is normalized as (mean : 0.5, std : 0.5) and reshaped to (height : 400), in contrast in training script (mean :0.59, std : 0.25) and reshaped to (height : 400).
Is this difference matter and do I have to change one of those input settings when testing our own dataset?

3. This is an extension of question 1. If the NDDS repository cannot be fixed, I am trying to make my own dataset by my hand :(. I am planning 500 labeled images for only 1 target object.
Do you have any insight whether 500 labeled images can make DOPE work fine for 1 target object?
This is time consuming work but still short of NDDS based dataset (which is 120K in paper)!

My questions are quite confusing. 
But I hope the promising answers from expert.
Thanks!

TontonTremblay, 2020-04-02:
1) @thangt Can you answer that one please? We looked at changing the repo to gitlab, but I am not sure what @thangt decided to do. 
2) In our testing we have not found much differences, but if you want to be certain, you can change these values easily. 
3) I never tried to play around with that little dataset, e.g., using 500 images. I do not think this would be enough.

yyc9268, 2020-04-03:
Thank you for comment.
I was luckily shared an NDDS from a generous researcher through dropbox.
Close the issue.

closed 2020-04-03 by yyc9268