NVlabs/Deep_Object_Pose #104: Is this code working with the latest CUDA 10  and cudnn 7.1?

astronaut71, 2020-04-09:
Hi
Im using Ubuntu 16.04, ROS kinetic  and having CUDA 10 and cudnn 7.10. Would be possible that this code is working with such configuration?

Thanks

TontonTremblay, 2020-04-10:
This looks like the same set up we have in our robot room, so I will think so.

The main problem you are going to have is that your block don't have the same sizes. DOPE uses PNP which implies that we know the size of objects a priori. If you have a changing size object with the same texture, you are going to need something to give you the size of that object, you could look into depth cameras for example. 

DOPE is build around the concept that we know the size of an object and its texture.

astronaut71, 2020-04-11:
OK. So if i wonna test it on a little big bigger size box, (just for test purposes to see if can get accurate pose), then where I should change the size of the box that gonna be detected? In some config file? Because i have some small boxes( 20cmx20cmx5cm). 

Thanks

TontonTremblay, 2020-04-11:
Yes it works on zed, but I have not tested with the depth estimation. We had some ideas with the zed camera, but we never really explored them. You logic seems correct for the box you want to detect.