NVlabs/Deep_Object_Pose #105: Build Deep Object Pose on AGX

chiehpower, 2020-04-10:
Is there any chance to build Deep Object Pose on AGX?
(AGX is one of Jetson devices.)

As I use JetPack 4.3 to flash the AGX, the os is Ubuntu 18.04.
However, I saw [this instructions](http://wiki.ros.org/kinetic/Installation/Ubuntu) which said only support Ubuntu 16.04. 
So I wonder is there any possible to build this Deep Object Pose on my device?

Thank you

TontonTremblay, 2020-04-10:
I tested on 18.04, I will update the readme to reflect that. We had someone internally that tested DOPE on one of the jetson device, the performance is not amazing (1-2 FPS from what I remeber), but it worked.

chiehpower, 2020-04-11:
OK Thank you for your information.
I wish the readme can also provide how to install on 18.04 from scratch. 
Thank you!

closed 2020-04-11 by chiehpower