NVlabs/Deep_Object_Pose #108: How to identify the hollow model better

xdw-123, 2020-04-14:
I've been able to identify my own training model. But the model is a kind of hollow structure, which requires too much light for the scene in the process of recognition. Is there a more general parameter or what are the requirements for synthetic data.

TontonTremblay, 2020-04-14:
Do you mind sharing some examples, like images?

xdw-123, 2020-04-15:
Hello, I have uploaded a group of synthesized data and physical pictures. I hope you have some suggestions. How to recognize any scene?URL：https://github.com/xdw-123/Deep_objcet_Pose_text-Image.git

TontonTremblay, 2020-04-15:
I did some testing on recognizing chairs with similar structure, e.g., thin legs. The network had no problem detecting the chair. I would think DOPE should work fine with your data (training + detection). Is it not the case?

xdw-123, 2020-04-16:
I think you are right. When you run rviz for simulation, there is frame dropping. How to speed up the recognition speed? My computer configuration: rtx8000, i9 7900