NVlabs/Deep_Object_Pose #109: what does values from depth frame represent ??

ravan786, 2020-04-16:
Hi,
Can someone please explain what does depth frame represents in FAT dataset?? The depth frame has three channel with RGB values(I guess!) How can i get something representing depth value from this RGB value?? And what does this depth values corresponds to?? Are they perpendicular distance from camera plane to all the points in the image??

I couldn’t find documentation for NDDS. Any help or documentation would be appreciated .
regards.

avinashsen707, 2020-04-17:
In FAT dataset , you can see the camera info file in each folders of dataset. Might be in the last rows .
So in there, you can see the camera parameters that are used by FAT authors in NDDS via UE4.

Basically there are Zed Camera, Realsense SR300 are available as default plugin in NDDS.
Others are neeededto be edited and added.

ravan786, 2020-04-18:
I saw the _camera_settings.json file but there i can’t find any setting or info about the depth frame.  Please help.

TontonTremblay, 2020-04-20:
I dont think you can modify the depth output. @thangt can you confirm this please. So what ever is written in the documentation, all depth frames should be similar.

UniqueVN, 2020-04-20:
@ravan786 
1. In the FAT dataset, the depth is captured in 16 bit grayscale image format, not 3 channel RGB.
2. Each pixel present the distance from the camera lense to the captured pixel in the normal image. That is how the depth camera work in the real world too. They are NOT perpendicular distance from the camera plane to the captured point.
3. Each of the pixel in the depth image have value from 0-65535 (16 bits). Its unit is decimilimeter or 0.1 milimeter or 10^-4 meter.
You can read more about the files format of the FAT dataset in its readme file here: https://research.nvidia.com/sites/default/files/pubs/2018-06_Falling-Things/readme_0.txt
4. In the NDDS, we support depth camera 8 bits and 16 bits too. If you use some prebuilt cameras that avinashsen707 mentioned above, they included the depth camera.

thangt, 2020-04-20:
@ravan786

In the FAT dataset, the depth is captured in 16 bit grayscale image format, not 3 channel RGB.
Each pixel present the distance from the camera lense to the captured pixel in the normal image. That is how the depth camera work in the real world too. They are NOT perpendicular distance from the camera plane to the captured point.
Each of the pixel in the depth image have value from 0-65535 (16 bits). Its unit is decimilimeter or 0.1 milimeter or 10^-4 meter.
You can read more about the files format of the FAT dataset in its readme file here: https://research.nvidia.com/sites/default/files/pubs/2018-06_Falling-Things/readme_0.txt
In the NDDS, we support depth camera 8 bits and 16 bits too. If you use some prebuilt cameras that avinashsen707 mentioned above, they included the depth camera.

mintar, 2020-04-21:
>  Each pixel present the distance from the camera lense to the captured pixel in the normal image. That is how the depth camera work in the real world too. They are NOT perpendicular distance from the camera plane to the captured point.

Sorry to disagree, but in a **depth image**, the pixels always represent the **perpendicular distance from the camera plane to the captured point**. This is the case for all depth cameras in the real world, and also for the output of NDSS and the FAT dataset. If the pixels represent the distance to the camera, that would be called a **range image**, not a depth image.

Before replying, I double-checked so I don't embarrass myself and add to the confusion. :)
If you want to follow it, here's how:

* Download some scene from the FAT dataset (that's where `000000.left.jpg`, `000000.left.depth.png` and `_camera_settings.json` below come from)
* `sudo apt install imagemagick pcl-tools`
* Convert the jpg to a png using imagemagick: `convert 000000.left.jpg 000000.left.png`
* From the `_camera_settings.json`, we see that `horizontal_fov = 64`. From the resolution (width = 960, height = 540), it follows that the vertical FOV is 540/960 * 64 = 36. So we can use these values to convert the depth image to a point cloud:
```bash
pcl_png2pcd --h_viewing_angle 64 --v_viewing_angle 36 --depth_unit mm --intensity_type UINT_8 000000.left.png 000000.left.depth.png 000000.pcd
```
* View the point cloud like this: `pcl_viewer 000000.pcd`. Result:
[image]
You can see that all planes are planar and not warped, so the conversion worked.

The part of the source code of `pcl_png2pcd` where the depth image is converted to a point cloud is here. The math is actually quite simple if the depth image is rectified:

https://github.com/PointCloudLibrary/pcl/blob/f934d1d2e0056ae04cc6b202b425ce6ae3f66e9d/tools/png2pcd.cpp#L117-L119

```c++
    x = (image_x - image_width / 2.0) / image_width * width;
    y = (image_height / 2.0 - image_y) / image_height * height;
    z = depth;
```

As you can see, the `z` value of the point cloud (in the camera frame) is identical to the depth value of the depth image. This proves my point that the depth pixels represent the perpendicular distance to the camera plane.

One note on the `pcl_png2pcd` conversion above: The resulting point cloud is not quite correct, because we specified `--depth_unit mm`. As @thangt pointed out, the depth unit is actually 10^-4 m, but `pcl_png2pcd` doesn't support that, so the resulting point cloud is 10x the actual size.