NVlabs/Deep_Object_Pose #110: can i get some real testing images for all the products ? It would be really helpful.

ravan786, 2020-04-18:


mintar, 2020-04-18:
Here you go: https://zenodo.org/record/2579172

ravan786, 2020-04-18:
thanks a lot man

ravan786, 2020-04-19:
@mintar I am sorry but I can’t resist to ask. Can you please tell me the reason for the QR code patterns on the table??

mintar, 2020-04-22:
The marker board on the table consists of ArUco markers. We used the marker board to estimate the camera pose in each frame. That way, we only had to label the ground truth poses of all objects in the scene once (in the global coordinate system defined by the marker board) and could then automatically transform them into the camera coordinate system for each frame.

By the way, I have an ICRA 2020 paper about the dataset. I still have to upload it to arXiV, but in the meantime, I can send it to you by email if you're interested.

ravan786, 2020-04-22:
> The marker board on the table consists of ArUco markers. We used the marker board to estimate the camera pose in each frame. That way, we only had to label the ground truth poses of all objects in [...]

@mintar that would be great! please mail it to krishna.ritesh94@gmail.com

closed 2020-04-24 by ravan786