NVlabs/Deep_Object_Pose #112: Is it possible to directly estimate poses from the bounding boxes?

syunni33, 2020-04-21:
Hi,

Currently, I haven't checked or applied for your work in my environment yet. 

However, before applying to mine, I want to know whether it is possible to directly estimate poses from the RGB images with bounding boxes as in an image below.

[image]

Is it possible to get the poses from the images that both RGB and bounding boxes are combined as an input, rather to use RGB images as you did?

TontonTremblay, 2020-04-21:
Yes you will get bounding boxes and poses like in the image.

syunni33, 2020-04-21:
It seems that I didn't explain well. I mean I want to use object bounding boxes as an input, not as an output.
Input: RGB image + object bounding boxes (bboxes are contained in the RGB images)
Output: pose

TontonTremblay, 2020-04-21:
You should be able to predict something, but I dont think it is going to be that precise. You wont be able to get the orientation, you can ruffly get the translation from the bbox.

closed 2020-04-21 by syunni33

PerceptMD, 2020-04-22:
I was thinking about cutting out the depth image to the predicted 2D bounding box size before trying to estimate any pose, as this would be less data to process.

@TontonTremblay why exactly will this approach not be able to get the orientation?

TontonTremblay, 2020-04-23:
Yeah you can for sure do that, I have seen system like this working without any problems.