NVlabs/Deep_Object_Pose #119: ROS visualization in RViz

Tingcccc, 2020-05-05:
Hello,

I am using  ROS Rviz to visualize  the weight I have trained. Currently, the bounding box can accurately mark the position of the target object, but the displayed cube will flicker and shake.
I would like to ask what method can be **avoid flickering**.
Look forward to your reply. Thank you!

TontonTremblay, 2020-05-05:
What do you mean flickering, do you have an example?

Tingcccc, 2020-05-06:
> What do you mean flickering, do you have an example?

Yes, the examples are like the video linked below. The cube overlap on the foot would flicker.

https://drive.google.com/file/d/1d5xAZ9hTwnzIsfWWPlGI4seUamUAFhVK/view?usp=sharing

https://drive.google.com/file/d/1fUiQs_ArX-snhzCJUEyEcv7Gol2zrAcY/view?usp=sharing

TontonTremblay, 2020-05-06:
This is very cool :P I am not sure about the overlay. The bounding boxes from DOPE does not flick. @mintar do you have an idea why the 3d mesh is flickering. 

As bonus questions, how did you train your network, did you use NDDS, do you get false positive on the other foot if you are not wearing your sandals?

Tingcccc, 2020-05-07:
> This is very cool :P I am not sure about the overlay. The bounding boxes from DOPE does not flick. @mintar do you have an idea why the 3d mesh is flickering.
>
> As bonus questions, how did you train [...]

Yes, I use NDDS to generate the synthetic data(DR) of right foot, and also add some real data to train the model. 
Indeed, in the state where the left foot is barefoot without wearing shoes, it will often get false positive. This problem is also a problem I have encountered now. For this situation, do you have any suggestions to solve it?

mintar, 2020-05-07:
The reason for the flickering is that we first publish a `MarkerArray` with a single marker and action `DELETEALL`:

https://github.com/NVlabs/Deep_Object_Pose/blob/685eaf901360f31b16602183d2a6539b074b2fbf/nodes/dope#L316

... and then we publish a second `MarkerArray` where we add all the real object markers for the current frame. The flickering occurs because the `DELETEALL` message deletes all markers (surprise!) and then we only add the others later.

The logic for marker visualization in RViz works something like this: A marker is uniquely identified by its `ns` (namespace) and `id` fields. It continues to be visualized in RViz until:

* another marker with the same `ns` and `id` is published (which will replace old the marker), or
* the marker's `lifetime` timeout is passed (in DOPE, we leave `lifetime` at its default value of 0, which means "forever"), or
* a `DELETEALL` marker is published.

The full documentation on the fields in the Marker message is [here](http://docs.ros.org/api/visualization_msgs/html/msg/Marker.html).

I have added a pull request (#120) that should hopefully fix this issue. Since I don't have access to the hardware at the moment, I couldn't test it though. @Tingcccc : Could you try if the pull request solves the flickering problem?

Cool project by the way! :)

Tingcccc, 2020-05-07:
> The reason for the flickering is that we first publish a `MarkerArray` with a single marker and action `DELETEALL`:
> 
> https://github.com/NVlabs/Deep_Object_Pose/blob/685eaf901360f31b16602183d2a6539b074b2fbf/nodes/dope#L316
> 
> ... and then we publish a second `MarkerArray` where we add all the real object markers for the current frame. The flickering occurs because the `DELETEALL` message deletes all markers (surprise!) and then we only add the others later.
> 
> The logic for marker visualization in RViz works something like this: A marker is uniquely identified by its `ns` (namespace) and `id` fields. It continues to be visualized in RViz until:
> 
> * another marker with the same `ns` and `id` is published (which will replace old the marker), or
> * the marker's `lifetime` timeout has passed (in DOPE, we leave `lifetime` at its default value of 0, which means "forever"), or
> * a `DELETE` marker with the same `ns` and `id` is published, or
> * a `DELETEALL` marker is published.
> 
> The full documentation on the fields in the Marker message is [here](http://docs.ros.org/api/visualization_msgs/html/msg/Marker.html).
> 
> I have added a pull request (#120) that should hopefully fix this issue. Since I don't have access to the hardware at the moment, I couldn't test it though. @Tingcccc : Could you try if the pull request solves the flickering problem?
> 
> Cool project by the way! :)

@mintar Thank you for your response, it was very useful and helpful! 
I changed marker.action = Marker.DELETEALL to Marker.DELETE  that the displayed object no longer flickers.

mintar, 2020-05-07:
>  I changed marker.action = Marker.DELETEALL to Marker.DELETE that the displayed object no longer flickers.

That will have the same effect as if you had removed the `DELETEALL` marker completely. `DELETE` only affects one specific marker, and only has an effect if the `id` and `ns` fields are filled out.

This means that if you just change `DELETEALL` to `DELETE` in the existing code, the flickering will be gone, **but if will not work** in the following situation:

* DOPE publishes objects 1, 2, 3, 4, 5
* after this, DOPE only publishes objects 1, 2, 3
* the markers for objects 4 and 5 will not be deleted and remain visible forever.

My PR #120 fixes this. Please try it instead. If you are on Ubuntu, you can try it like this:

```bash
sudo apt install snapd
sudo snap install hub --classic
cd <your DOPE directory>
hub pr checkout 120
```

TontonTremblay, 2020-05-07:
@Tingcccc if you have the capacity to test the PR, it would be fantastic you did so. 

As for the left and right foot, you are going to have to show the network examples of the left foot and right foot and only label one of them. This is going to be a hard problem, but the idea is to curate your training data to make sure that the left-right foot structure is there and label them correctly. But there are cases that are going to be confusing, but I am sure you can get it there 95% of the time.

Tingcccc, 2020-05-08:
> @Tingcccc if you have the capacity to test the PR, it would be fantastic you did so.
>
> As for the left and right foot, you are going to have to show the network examples of the left foot and right [...]

Thank you for your suggestion, I will try to use this method to generate data. 
In addition, if there are real images added, will it have the same effect as PR synthetic data?

TontonTremblay, 2020-05-09:
PR is pull request on github. Feel free to explore using real data, I dont have much experience with them for that problem.

Tingcccc, 2020-05-14:
@mintar Hello, I have an additional question, how do I display the 3D cube and the bounding box on the same image?
Thank you.

mintar, 2020-05-18:
> @mintar Hello, I have an additional question, how do I display the 3D cube and the bounding box on the same image?

Do you mean the yellow 3D bounding box from your video? That is the same as the 3D cube, so why would you want to visualize both?

At the moment, you can only visualize the 3D mesh and the 3D bounding box by using the "Camera" display in RViz. You cannot easily use the "Camera" display for the `dope/rgb_points` topic (the one with the yellow bounding box) because it doesn't have an associated `camera_info` topic; you have to use the "Image" display instead. If you really want to do this, you have to make sure that there is a `dope/camera_info` topic. For example, you could use [topic_tools/relay](http://wiki.ros.org/topic_tools/relay) to republish the camera_info topic from your camera to the `dope/camera_info` topic. But again, why do you want to do this?

Tingcccc, 2020-05-18:
> > @mintar您好，我還有一個問題，如何在同一張圖片上顯示3D立方體和邊界框？
> 
> 您是說視頻中的黃色3D邊框嗎？這與3D立方體相同，那麼為什麼要同時可視化它們呢？
> 
> 目前，您只能使用RViz中的“攝像機”顯示來可視化3D網格和3D邊界框。您無法輕鬆使用`dope/rgb_points`主題的“攝像機”顯示（帶有黃色邊框的顯示），因為它沒有相關的`camera_info`主題；您必須改為使用“圖像”顯示。如果您確實要執行此操作，則必須確保有一個`dope/camera_info`主題。例如，您可以使用[topic_tools / relay](http://wiki.ros.org/topic_tools/relay)將camera_info主題從相機重新發佈到該`dope/camera_info`主題。但是，您又為什麼要這樣做呢？

Because I want to generate the effect of marker.mesh_resource overlaying on the Image, but found that the marker directly stacked on the camera output will be delayed, not synchronized. 
It is speculated that due to Import data into the model and do predictions will make "rgb_point" slower than "camera_raw", so if I can superimpose the marker on the Image (rgb_point), the movement of the marker should be completely synchronized with the image. is it right?
Could I use topic_tools / relay to achieve such an effect?

mintar, 2020-05-18:
Ah, I get it. Yes, you are right: image_raw is published immediately when the camera records an image, and both the markers and the "rgb_point" topic are published at the same time *after* computing the predictions, so there will be a visible delay between image_raw and the markers.

I have just checked: My suggestion of simply using "topic_tools/relay" will not work, because we don't properly set a time stamp in the header of the `rgb_points` message. 

I have just added pull request #121 , which should resolve this issue, but I haven't tested it yet. @Tingcccc , please try it out. Instructions how to do so are in my [previous comment](https://github.com/NVlabs/Deep_Object_Pose/issues/119#issuecomment-625235432).

Tingcccc, 2020-05-19:
After I modified the code according to your method, it is indeed executable. Thanks a lot.

PR #120 (2020-09-22): Fix markers flickering in RViz
Fixes #119 .

This pull request is not tested yet because I don't have access to the hardware at the moment. Please test before merging.
commit 33d771e0 (2020-09-22): Fix markers flickering in RViz

Fixes #119
closed 2020-09-22 by TontonTremblay