NVlabs/Deep_Object_Pose #122: Unable to install ros-kinetic-camera-info-manager-py

Avi71, 2020-05-25:
When I run the command "rosdep install --from-paths src -i --rosdistro kinetic" I get the following - 

executing command [sudo -H apt-get install ros-kinetic-camera-info-manager-py]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  linux-headers-4.15.0-29 linux-headers-4.15.0-29-generic
  linux-headers-4.15.0-36 linux-headers-4.15.0-36-generic
  linux-image-4.15.0-29-generic linux-image-4.15.0-36-generic
  linux-modules-4.15.0-29-generic linux-modules-4.15.0-36-generic
  linux-modules-extra-4.15.0-29-generic linux-modules-extra-4.15.0-36-generic
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  ros-kinetic-camera-info-manager-py
0 upgraded, 1 newly installed, 0 to remove and 341 not upgraded.
Need to get 14.5 kB of archives.
After this operation, 79.9 kB of additional disk space will be used.
Err:1 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-camera-info-manager-py amd64 0.2.3-0xenial-20181107-012231-0800
  404  Not Found [IP: 64.50.236.52 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-camera-info-manager-py/ros-kinetic-camera-info-manager-py_0.2.3-0xenial-20181107-012231-0800_amd64.deb  404  Not Found [IP: 64.50.236.52 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install ros-kinetic-camera-info-manager-py] failed

Can anyone please help me install it?

mintar, 2020-05-25:
This has nothing to do with DOPE; it's probably a transient error. Please just try again. If you still have problems, ask over at https://answers.ros.org/.

TontonTremblay, 2020-05-26:
Looks like your apt-get is set up on the old addresses for ROS packages.

Avi71, 2020-05-31:
Hey, this issue has been solved. When I was using the command "sudo apt update" I got the following
 "Hit:1 http://in.archive.ubuntu.com/ubuntu xenial InRelease
Hit:2 http://in.archive.ubuntu.com/ubuntu xenial-updates InRelease                            
Hit:3 http://in.archive.ubuntu.com/ubuntu xenial-backports InRelease                          
Hit:5 http://ppa.launchpad.net/js-reynaud/kicad-4/ubuntu xenial InRelease                     
Get:4 http://packages.ros.org/ros/ubuntu xenial InRelease [4,678 B] 
Hit:6 http://security.ubuntu.com/ubuntu xenial-security InRelease        
Err:4 http://packages.ros.org/ros/ubuntu xenial InRelease                     
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
Hit:7 http://ppa.launchpad.net/obsproject/obs-studio/ubuntu xenial InRelease  
Fetched 4,678 B in 1s (2,769 B/s)                 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
350 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://packages.ros.org/ros/ubuntu xenial InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
W: Failed to fetch http://packages.ros.org/ros/ubuntu/dists/xenial/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
W: Some index files failed to download. They have been ignored, or old ones used instead.
"

Then I found that when running the command "rosdep install --from-paths src -i --rosdistro kinetic" I used to get the following - 
"executing command [sudo -H apt-get install ros-kinetic-camera-info-manager-py]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  linux-headers-4.15.0-29 linux-headers-4.15.0-29-generic
  linux-headers-4.15.0-36 linux-headers-4.15.0-36-generic
  linux-image-4.15.0-29-generic linux-image-4.15.0-36-generic
  linux-modules-4.15.0-29-generic linux-modules-4.15.0-36-generic
  linux-modules-extra-4.15.0-29-generic linux-modules-extra-4.15.0-36-generic
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  ros-kinetic-camera-info-manager-py
0 upgraded, 1 newly installed, 0 to remove and 350 not upgraded.
Need to get 14.5 kB of archives.
After this operation, 79.9 kB of additional disk space will be used.
Err:1 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-camera-info-manager-py amd64 0.2.3-0xenial-20181107-012231-0800
  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-camera-info-manager-py/ros-kinetic-camera-info-manager-py_0.2.3-0xenial-20181107-012231-0800_amd64.deb  404  Not Found [IP: 64.50.233.100 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install ros-kinetic-camera-info-manager-py] failed
"
In both the situations the common thing was ubuntu xenial, then I found out my key was old from the above errors which was preventing the packages to install. After updating the key everything is working fine. For anyone who gets the same error, to get the new key, check the following link - [https://answers.ros.org/question/325039/apt-update-fails-cannot-install-pkgs-key-not-working/](url)

closed 2020-05-31 by Avi71