NVlabs/Deep_Object_Pose #123: Using depth only

codesieve, 2020-05-26:
Hello,
I want to learn the object poses of my own objects with DOPE for a robot grasping task I have in my research. I have already managed to adept NDDS in a way where I am able to generate random images with these objects that look very much like the real images I get from my sensor. However, my Sensor only gives me depth images (and point clouds alternatively, but this would be harder to implement in NDDS just to start with, I guess). This is why I exported depth images and the feature json files only.

Now I am wondering: Does DOPE support using depth images only to learn, without RGB images? I think it can work as depth images have all the information needed for a solid pose estimation but I wanted to know your opponions on that before I start experimenting. I guess that I will have to change or adept the camera node as well as the camera config file.

But what do you think? Will it probably work? And can you give me hints on where to start with adaptations if needed?

Thank you so much in advance, I appreciate any help!

TontonTremblay, 2020-05-26:
NDDS exports depth, but I never personally tested how it holds compared to real depth sensors or how to augment it to look like a specific sensor. 

As to use DOPE on depth images, this is a very interesting question. I am not sure, depth images/point cloud are an interesting beast to play with. I don't have a specific opinion about it, but maybe you would like to look into DenseFusion. My two cents; 1) you need think about the input to the network 2) what the output is. 3) What do you regress too? Depth has been used to do pose estimation before, have you seen bayesian tracking or Simtrack? You might want to check them out before going down the road of investigating a new depth pose estimator if this is not your main research topic. Otherwise go in full steam.

codesieve, 2020-05-26:
Thanks for this assessment of the situation with DOPE and NDDS and the hints with Bayesian Tracking and Simtrack. I am going to check them out and decide on how to deal with my problem afterwards :)