NVlabs/Deep_Object_Pose #126: show the real FPS using rostopic

ROSNovice, 2020-06-20:
Hi Guys, do you know how to calculate the average frame per second in topic (dope_webcam/rgb_points).
Did you use rostopic hz to measure the fps of dope detection? i used it but i am not quite sure with the result, it is only 2,5 for average fps, for 1 object detection. For 3 object detection, it showed 0.8 for average frames per second (my GPU is RTX 2070)
please kindly help me the best method to calculate the FPS

mintar, 2020-06-20:
rostopic hz should work, that's what I use.

ROSNovice, 2020-06-21:
> rostopic hz should work, that's what I use.

this is my result
https://youtu.be/9k2b9KGPpiI

i am not sure it is only 2.4 Average FPS for detecting one object. Do you have any ideas or suggestions? thanks a lot for your reply

mintar, 2020-06-22:
I just counted the times when each frame changed in your video and came up with 100 frames. Divide that by 42 seconds video, and you come up with 2.38 hz, which is exactly what `rostopic hz` says.

mintar, 2020-06-22:
Also, if you look at [this post](https://github.com/NVlabs/Deep_Object_Pose/issues/54#issuecomment-575404071) by @TontonTremblay (one of DOPE's authors), you can see that he gets around 3.2 Hz on a 2080ti, so around the same performance as yours. You can make it faster (but less precise!) by reducing the `stop_at_stage` parameter:

https://github.com/NVlabs/Deep_Object_Pose/blob/65a657ada41c2832334cbb09ad75e5d11364f521/src/dope/inference/detector.py#L41

ROSNovice, 2020-07-14:
hi @mintar, i saw in another issue and discussion, @TontonTremblay get 10-12 fps, how he calculate the fps? i do not understand

mintar, 2020-07-14:
In @TontonTremblay's  video, I just counted the frames and divided by the video length to come up with 3.2 Hz. But in general, it's better to just use `rostopic hz /dope/detected_objects`.

closed 2020-09-02 by nv-jeff