NVlabs/Deep_Object_Pose #127: Training Times and Errors on the Way

Ademord, 2020-06-23:
I have been trying to train on a single object, currently my command is:

```python train.py --data /root/catkin_ws/src/dope/fat --object 037_scissors_16k --outf 037_scissors_16k --batchsize 4 --workers 4 --gpuids 0 1 2 3```

but I keep running out of memory with the error

```RuntimeError: DataLoader worker (pid 54) is killed by signal: Bus error. It is possible that dataloader's workers are out of shared memory. Please try to raise your shared memory limit.```

So i increased my VM's memory and also edited the run_docker.sh by adding ```-m 32G --memory-swap -1 --shm-size 32g```. **Is this ok?**

Then my training was getting stuck after "Training from pretrained network vgg", so at some point before I had run into this problem. I ended up force-reinstall --upgrading ```torch``` and **now finally** it's training.
As of the command above, I am running 8 default workers and a batch size 16 on 4 GPUs... and one epoch has taken me 5h approximately. Could someone that has trained, give me feedback if that's OK training time? it's looking like a week of training at this pace... since the default is 60 epochs...

TontonTremblay, 2020-06-27:
I never used a VM to train neural network, I am sorry my help is limited here. but a batchsize of 16 for multiple t4 looks very low. I think on 4 p100 I run 96 batchsize. Can you check if there is `pin_memory=True` in the data loader that might have impact. My normal training time is 2 days on 4 p100.

Ademord, 2020-06-30:
Hello!
Thanks a lot for your reply! I will check with pin_memory.

I ended up being able to push the batch size to 150 after increasing my shm-size (32GB).
My training time was 7 days.

I have successfully visualized the banana from the FAT dataset
[image]

but I dont understand now 2 things. In the config, i used the "dimensions" of "bleach" (changed to "banana") because I dont know where to get these dimension values? 
Also the arrow on the right side of rviz seems a bit huge, must I tweak the "tranform" in config? 

Sorry that I am lost, I could use some guidance on these last steps :+1:

TontonTremblay, 2020-07-01:
Yeah you to measure your banana's cuboid dimension. If it is the toy, just look at the dimensions on the 3d model, e.g., check the aabb dimensions. Also the detection you share does not look too good, if you only trained on FAT dataset, I would not be surprised that the sim-2-real is quite limited. Can you show us a picture from `dope_banana_rgb_points` or something like this?

mintar, 2020-07-01:
> but I dont understand now 2 things. In the config, i used the "dimensions" of "bleach" (changed to "banana") because I dont know where to get these dimension values?

It is absolutely **crucial** to use the correct dimensions, otherwise the solvePnP algorithm will not work; this is *not* just a "nice to have" for visualization purposes.

You can find the correct values in the `_object_settings.json` file from the FAT dataset:

```json
		{
			"class": "011_banana_16k",
			"cuboid_dimensions": [ 19.717399597167969, 3.8649001121520996, 7.4065999984741211 ]
		},
```

> Also the arrow on the right side of rviz seems a bit huge, must I tweak the "tranform" in config?

No, you can configure the "Pose" display in RViz; by default, it's a 1 meter long arrow, but you can change the size of the arrow, or (more usefully) change it to the "axes" visualization.

Some more replies to your mail:

> I have trained 39 epochs (server restarted and couldnt finish all 60, woops) on the banana from the FAT dataset.

That's okay, the difference between 39 and 60 epochs isn't that big. Also, it is possible to resume the training if you want, you don't have to start over!

> I am a bit confused about the config parameters... i didnt know what to use on the "dimensions", "model_transformations", and "meshes".

* dimensions: see above
* meshes: This is purely for visualization, but very useful. You can download the "banana" mesh using [nvdu_ycb](https://github.com/NVIDIA/Dataset_Utilities#nvdu_ycb). If you put the path to that mesh into the "meshes" config, RViz will show mesh markers for it.
* model_transforms: you don't need this for the FAT dataset, because the orientation is the same in the training data (from UE4) and in the mesh model. For some of my models, the axes were messed up somewhere between exporting from blender and importing into UE4. If you notice that there is a constant 90° rotation error in the pose returned by DOPE, e.g. the x axis should be z, -y should be y and z should be x, you can put in a model_transform like this:

```yaml
model_transforms: {
    "gear_box_mount":            [[ 0,  0,  1,  0],
                                  [ 0, -1,  0,  0],
                                  [ 1,  0,  0,  0],
                                  [ 0,  0,  0,  1]],
}
```

But again, don't do this for the FAT models, they are already correct.

> I am a bit confused, why there isnt any overlay on the top

Maybe you used the "Image" display in RViz? Only the "Camera" display has overlays.

Ademord, 2020-07-01:
Wow thanks for the thorough answer!! 😁
How do I continue training from where it left off?.. I didn't know how to do that :/

mintar, 2020-07-01:
Like this:

```bash
python train.py --net /root/dope_models/my_object/net_epoch_10.pth
```

https://github.com/NVlabs/Deep_Object_Pose/blob/65a657ada41c2832334cbb09ad75e5d11364f521/scripts/train.py#L1124-L1126

Ademord, 2020-07-01:
Hello
Result looks like this now :)
[image]

[image]

Thank u guys!

closed 2020-07-02 by TontonTremblay

DieuwertW, 2021-01-06:
Just a quick question, I'm trying to use Google Colab to train, so I'd want to continue training after the session is done. Now I was trying to use :

> ```shell
> python train.py --net /root/dope_models/my_object/net_epoch_10.pth
> ```

But when doing this it will just copy the epoch 60 times in a minute or so. Not really the result I was hoping to get. Does anyone have a solution for that?
I am trying to do so:

!python Deep_Object_Pose/scripts/train.py --net /mydrive/DOPE/net_epoch_1.pth --data data/training/PersistentMapTest --object Objctname --outf /mydrive/DOPE/test --epochs 2

Does anyone know if this will work? Thanks in advance!