NVlabs/Deep_Object_Pose #128: Stack of cubes not getting detected

Vaibhavjolly, 2020-06-24:
Hi sir,

I have created Cubes Dataset(30k images) using NDDS tool . I was trying to test the model with synthetic data as well as real data . Multiple Cubes are detected when they are isolated and on the floor (synthetic ones) but are not detected when multiple cubes are stacked . 
I want to detect the cubes when they are stacked but I can't detect those. I could not find the reason for that as I already have considered those images in my training set(multiple cubes stacked) .Can you please guide me on this ?Even the top view cubes are not detected .
Also the accuracy with the real data is very less .How can I improve that ?

Thanks,
Vaibhav

TontonTremblay, 2020-06-27:
Could you share some images of you training data? If you want to detect cubes detected on a tower formation you need to generate images of cube tower. I have a paper at icra 2018 on cube stacking detection using something similar to DOPE. Most likely the problem is on the training data. Are you interested in any cube or a specific color cube?

Vaibhavjolly, 2020-07-01:
Hi @TontonTremblay , Yeah sure I am attaching few training images .I already have considered cube tower data . Can you please also share that paper link similar to dope for cube stacking detection , and if there is any open source code . I am interested in cubes with different colors (red,green,yellow) stacking .
Thanks for your help !
[image]
[image]

TontonTremblay, 2020-07-02:
I am sorry we did not open sourced the paper, and yes that was the one I was referring to. Could you show me the heat maps on the different testing images. Sorry for asking more information.

closed 2025-08-08 by nv-jeff