NVlabs/Deep_Object_Pose #132: How to calculate `pose_transform_permuted`

belorenz, 2020-07-06:
I'am creating my own FAT-like dataset. Therefore I would need to calculate the pose_transform_permuted. As described in FAT dataset overview the pose_transform_permuted matrix is the following:

>   - 4x4 transformation (redundant, can be computed from previous) (`pose_transform_permuted`)

> *Note:*  Like the `fixed_model_transform`, the `pose_transform_permuted` is actually the transpose of the matrix.  Moreover, after transposing, the columns are permuted, and there is a sign flip (due to UE4's use of a lefthand coordinate system).  Specifically, if `A` is the matrix given by `pose_transform_permuted`, then actual transform is given by `A^T * P`, where `^T` denotes transpose, `*` denotes matrix multiplication, and the permutation matrix `P` is given by

How do I calculate this matrix from the quaternion / location I already have?

mintar, 2020-07-07:
I had the same problem when I was recording my own FAT-like dataset, [YCB-M](https://zenodo.org/record/2579172). I have created a set of tools for creating the annotations. Your comment prompted me to finally put it on GitHub:

https://github.com/mintar/ycb_multicam_dataset_tools

It's a still a bit rough around the edges, but maybe you'll find it useful.

There's some documentation about the `pose_transform_permuted` here:

https://github.com/mintar/ycb_multicam_dataset_tools/blob/3249dffdb77527871d20f92b870d1e1de0bf9cc5/make_segmentation_imgs.py#L441-L465

And a function to generate it is here:

https://github.com/mintar/ycb_multicam_dataset_tools/blob/3249dffdb77527871d20f92b870d1e1de0bf9cc5/make_segmentation_imgs.py#L468-L482

To be honest, I think that `pose_transform_permuted` is not really used anywhere (neither in DOPE nor NVDU); you could probably just leave it out.

belorenz, 2020-07-27:
Thanks for the hints!

closed 2020-07-27 by belorenz