NVlabs/Deep_Object_Pose #133: Training loss and domain randomization problem

IbrahimMCode, 2020-07-07:
Hello,

I'm training on a new object and i already had a lot of datasets for this object.
I already tried a lot of photorealistic only datasets and domain randomization ones, also some mixed datasets.
My problem is that whenever i add domain randomization images to my training i have no detection on real images.
While trying on 40k or 60k photorealistic images only the detection performed well on real images.
I already tried 40k photorealistic + 10k domain randomization also a 60k / 60k dataset etc. and always the DR Images are causing a problem.

What made me confused was training the drill object using the FAT Dataset and 60k DR Images(found this on one of the issues).

Is there a dummy domain randomization dataset generated or can you provide me with a small example about the drill generated images?

Also i'm having a problem with the training loss it starts for example from 0.06 and ends at 0.007.I already tried different GPUs combination (8 V100 or 2 Titan RTX) and it's the same problem. Do you have any idea about it?

Thank you in advance.

TontonTremblay, 2020-07-08:
Which object are you trying to train for? The drill? If so I will upload a dataset on google drive for you. That loss looks pretty high. I normally get around 0.0004 or 0.0003. So my guess it would have to do with your training data. But I am really not sure. I have used the training code here on different occasion and it works fairly well although it is different the one I use internally now.

IbrahimMCode, 2020-07-08:
Yes. the drill,
I would be thankful if you can provide me with a link for the dataset.

TontonTremblay, 2020-07-09:
One of the problem with the drill and doing cuboid is that 2 points fall in the middle of nowhere. I think this is hurtful for the pose estimation, so that might be a problem.

IbrahimMCode, 2020-07-09:
Yes, it's very dependent of the background.

closed 2025-08-15 by nv-jeff