NVlabs/Deep_Object_Pose #134: How to display images like the example

ZZ-111, 2020-07-17:
First of all, thank you for this great open source project.
My questions are:
How to display images like the example(coordinates. boxes.mask in same image):
[image]
I can only show them separately like this:
[image]
If the camera subscribes image topic /dope/rgb_points ,it shows like this:
[image]

ZZ-111, 2020-07-22:
[image]
I modified some code, which now display borders and coordinates, but how to make them exactly the same as the sample pictures (borders, color coordinates, mask).

mintar, 2020-07-23:
I think the sample pictures were generated using [NVDU](https://github.com/NVIDIA/Dataset_Utilities).

Your RViz screenshot already looks very good. What do you want to do differently?

* Remove the text? That's possible by disabling some namespace in the "MarkerArray" Display.
* Make the coordinate axes more colorful? Then you'll have to rotate the 3D view in RViz so that you look at the object from roughly the same direction as the camera. This is because RViz renders a light behind the camera, and in your screenshot you look the opposite direction, so that the coordinate axes are in the shadow (they are lighted from the opposite side).

ZZ-111, 2020-07-23:
> I think the sample pictures were generated using [NVDU](https://github.com/NVIDIA/Dataset_Utilities).
>
> Your RViz screenshot already looks very good. What do you want to do differently?
>
> * Remove the [...]

Thank you for your help. Now the coordinate system is colorful!
[image]

Another problem is that if i want to make the picture exactly the same as the sample picture,do I need to use NVDU?

mintar, 2020-07-23:
What do you mean by "exactly the same"?

ZZ-111, 2020-07-23:
> What do you mean by "exactly the same"?

I can see that the coordinates on this graph are not fully displayed, and some of them are hidden.
[image]

mintar, 2020-07-23:
I think you will have to use NVDU for this. But then you'll have to write some code to export the detections to NVDU format.

ZZ-111, 2020-07-23:
> 我认为您必须为此使用NVDU。但是随后您必须编写一些代码以将检测结果导出为NVDU格式。

Ok.Thank you!

ZZ-111, 2020-07-23:
> I think you will have to use NVDU for this. But then you'll have to write some code to export the detections to NVDU format.

Ok.Thank you!

closed 2020-07-24 by ZZ-111