NVlabs/Deep_Object_Pose #140: cv2 error: DLT algorithm needs at least 6 points for pose estimation

sebastian-ruiz, 2020-09-24:
I am running DOPE in the docker container with only the soup can active in the weights in the config_pose.yaml. The usb camera is a logitech c270 and it has been calibrated following the guide.

The moment that the camera has the soup can in view, dope crashes, with the following error:

```
root@rotterdam:/# roslaunch dope dope.launch
... logging to /root/.ros/log/6a198ad8-fe65-11ea-ad39-a4bb6d45a510/roslaunch-rotterdam-706.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://rotterdam:46331/

SUMMARY
========

CLEAR PARAMETERS
 * /dope/

PARAMETERS
 * /dope/class_ids/AlphabetSoup: 9
 * etc...
 * /dope/class_ids/soup: 5
 * /dope/class_ids/sugar: 6
 * /dope/dimensions/AlphabetSoup: [8.35550022125244...
 * etc...
 * /dope/dimensions/soup: [6.76593780517578...
 * /dope/dimensions/sugar: [9.26773071289062...
 * /dope/downscale_height: 500
 * etc...
 * /dope/draw_colors/soup: [255, 101, 0]
 * /dope/draw_colors/sugar: [232, 222, 12]
 * /dope/input_is_rectified: True
 * etc...
 * /dope/mesh_scales/soup: 0.01
 * /dope/mesh_scales/sugar: 0.01
 * /dope/sigma: 3
 * /dope/thresh_angle: 0.5
 * /dope/thresh_map: 0.01
 * /dope/thresh_points: 0.1
 * /dope/topic_camera: /usb_cam/image_raw
 * /dope/topic_camera_info: /usb_cam/camera_info
 * /dope/topic_publishing: dope
 * /dope/weights/soup: package://dope/we...
 * /rosdistro: noetic
 * /rosversion: 1.15.8

NODES
  /
    dope (dope/dope)

ROS_MASTER_URI=http://localhost:11311

process[dope-1]: started with pid [721]
Loading DOPE model '/root/catkin_ws/src/dope/weights/soup_60.pth'...
    Model loaded in 3.0019054412841797 seconds.
Running DOPE...  (Listening to camera topic: '/usb_cam/image_raw')
Ctrl-C to stop
[ERROR] [1600958723.403318]: bad callback: <bound method Subscriber.callback of <message_filters.Subscriber object at 0x7f30ac10a220>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 76, in callback
    self.signalMessage(msg)
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 58, in signalMessage
    cb(*(msg + args))
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 240, in add
    self.signalMessage(*msgs)
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 58, in signalMessage
    cb(*(msg + args))
  File "/root/catkin_ws/src/dope/nodes/dope", line 247, in image_callback
    results = ObjectDetector.detect_object_in_image(
  File "/root/catkin_ws/src/dope/src/dope/inference/detector.py", line 256, in detect_object_in_image
    detected_objects = ObjectDetector.find_object_poses(vertex2, aff, pnp_solver, config)
  File "/root/catkin_ws/src/dope/src/dope/inference/detector.py", line 273, in find_object_poses
    location, quaternion, projected_points = pnp_solver.solve_pnp(points)
  File "/root/catkin_ws/src/dope/src/dope/inference/cuboid_pnp_solver.py", line 87, in solve_pnp
    ret, rvec, tvec = cv2.solvePnP(
cv2.error: OpenCV(4.4.0) /tmp/pip-req-build-6179nsls/opencv/modules/calib3d/src/calibration.cpp:1173: error: (-2:Unspecified error) in function 'void cvFindExtrinsicCameraParams2(const CvMat*, const CvMat*, const CvMat*, const CvMat*, CvMat*, CvMat*, int)'
> DLT algorithm needs at least 6 points for pose estimation from 3D-2D point correspondences. (expected: 'count >= 6'), where
>     'count' is 4
> must be greater than or equal to
>     '6' is 6
```

sebastian-ruiz, 2020-09-24:
Testing with the soup can on the Kinetic version works fine.

TontonTremblay, 2020-09-24:
This is a new behaviour from opencv. @nv-jeff can you look into this change please, or maybe have the version of cv2 in the docker image to be on an earlier version.

sbirchfield, 2020-09-24:
@sebastian-ruiz If you change the flag `pnp_algorithm` (not exposed at the high level, but see line 45 of cuboid_pnp_solver.py), do you get different behavior?  Different algorithms require different minimum number of points.

nv-jeff, 2020-09-24:
Thanks for alerting us. There was a bug in how we handled different versions of OpenCV that has now been fixed. Please pull and try again. I will leave this issue open for now, until I hear from you that it is fixed.

sebastian-ruiz, 2020-09-25:
I deleted the docker container and the old docker image, pulled the update and rebuilt the docker image. I am still getting the same/very similar error to do with cv2. 

Is it supposed to be running OpenCV(4.4.0)? 

Thank you for your help.

```
➜ ./run_dope_docker.sh 
Container name     : nvidia-dope-v2
Host directory     : /home/sebastian/docker/dope
Container directory: /root/catkin_ws/src/dope
Found DOPE docker container: 1d5cc976fe0a.
Found running nvidia-dope-v2 container, attaching bash...
root@rotterdam:/# roslaunch dope dope.launch
... logging to /root/.ros/log/726c9346-ff38-11ea-86f0-a4bb6d45a510/roslaunch-rotterdam-1281.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://rotterdam:37897/

SUMMARY
========

..... (omitted) ...

ROS_MASTER_URI=http://localhost:11311

process[dope-1]: started with pid [1606]
Loading DOPE model '/root/catkin_ws/src/dope/weights/soup_60.pth'...
    Model loaded in 3.988426446914673 seconds.
Running DOPE...  (Listening to camera topic: '/usb_cam/image_rect_color')
Ctrl-C to stop
[ERROR] [1601044108.540632]: bad callback: <bound method Subscriber.callback of <message_filters.Subscriber object at 0x7fb3cc220250>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 76, in callback
    self.signalMessage(msg)
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 58, in signalMessage
    cb(*(msg + args))
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 240, in add
    self.signalMessage(*msgs)
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 58, in signalMessage
    cb(*(msg + args))
  File "/root/catkin_ws/src/dope/nodes/dope", line 254, in image_callback
    results = ObjectDetector.detect_object_in_image(
  File "/root/catkin_ws/src/dope/src/dope/inference/detector.py", line 256, in detect_object_in_image
    detected_objects = ObjectDetector.find_object_poses(vertex2, aff, pnp_solver, config)
  File "/root/catkin_ws/src/dope/src/dope/inference/detector.py", line 273, in find_object_poses
    location, quaternion, projected_points = pnp_solver.solve_pnp(points)
  File "/root/catkin_ws/src/dope/src/dope/inference/cuboid_pnp_solver.py", line 89, in solve_pnp
    ret, rvec, tvec = cv2.solvePnP(
cv2.error: OpenCV(4.4.0) /tmp/pip-req-build-vu_aq9yd/opencv/modules/calib3d/src/calibration.cpp:1173: error: (-2:Unspecified error) in function 'void cvFindExtrinsicCameraParams2(const CvMat*, const CvMat*, const CvMat*, const CvMat*, CvMat*, CvMat*, int)'
> DLT algorithm needs at least 6 points for pose estimation from 3D-2D point correspondences. (expected: 'count >= 6'), where
>     'count' is 4
> must be greater than or equal to
>     '6' is 6
```

nv-jeff, 2020-09-25:
Can you tell me what lines 53-62 in cuboid_pnp_solver.py look like in your codebase? It should look like:
`
            if CuboidPNPSolver.cv2majorversion == 2:
                pnp_algorithm = cv2.CV_ITERATIVE
            elif CuboidPNPSolver.cv2majorversion > 2:
                pnp_algorithm = cv2.SOLVEPNP_ITERATIVE
                # Alternative algorithms:
                # pnp_algorithm = SOLVE_PNP_P3P
                # pnp_algorithm = SOLVE_PNP_EPNP
            else:
                assert True, "DOPE will not work with versions of OpenCV earlier than 2.0"
`
If thi s is the case, it's possible that DOPE is failing to find enough points in your input image to construct a bounding box. Do you have a sample image that is failing for you that I could take a look at?

sbirchfield, 2020-09-25:
cv2.SOLVEPNP_ITERATIVE requires at least 6 points.  This can be seen here:  https://github.com/opencv/opencv/blob/f6ec0cd827310ba801575f846720da4433843e50/modules/calib3d/src/solvepnp.cpp#L820

If you have fewer than 6 points, you should use a different algorithm, like cv2.SOLVEPNP_EPNP .  These other algorithms require a minimum of 4 points, I believe.

sebastian-ruiz, 2020-09-28:
> Can you tell me what lines 53-62 in cuboid_pnp_solver.py look like in your codebase?

Yes those are exactly the lines I have.

> If this is the case, it's possible that DOPE is failing to find enough points in your input image to construct a bounding box. Do you have a sample image that is failing for you that I could take a look at?

I tried to record where this happens (see here: https://imgur.com/a/m67zWG7) but all that you see is that DOPE freezes all of a sudden, with the accompanying terminal output the same as in my previous comment.

sebastian-ruiz, 2020-09-28:
I recorded the same using the old ROS kinetic version and here DOPE doesn't crash, however sometimes the bounding box does show as only a single line or a dot. This could be where the Noetic version is crashing. Here is the video of the kinetic version: https://imgur.com/a/YBEWj1t

nv-jeff, 2020-09-28:
Could you provide us with a single image that is failing for you? I've been unable to reproduce this locally, but I've seen other reports that OpenCV 4.4 may have regressions in its perspective n-point code, and I'd like to chase this issue down.

sebastian-ruiz, 2020-09-29:
It seems to fail with the soup cans when the soup cans are only partly in the frame. The soup can weights are from the googlde drive link given in the readme.md. For example it crashes on this image:

[image]

I am using a logitech c270 camera:
```
    "width": 640,
    "height": 480,
    "fx": 819.614890,
    "fy": 825.737837,
    "cx": 293.182081,
    "cy": 250.299296
```

or in the yaml format for ROS:
```
image_width: 640
image_height: 480
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [819.614890, 0.000000, 293.182081, 0.000000, 825.737837, 250.299296, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.043100, 0.162156, 0.000011, -0.005394, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [818.209778, 0.000000, 291.093218, 0.000000, 0.000000, 827.370850, 250.259915, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]
```

Ademord, 2020-10-01:
I got the same error yesterday when running dope-python3 (my own model.pth testing on a rosbag).

nv-jeff, 2020-10-01:
You've both encountered a regression of sorts in OpenCV, where it asserts if insufficient points are provided to the solvePnP function. Previously, it would fall back to another algorithm.  I have updated our logic to handle this. You don't need to rebuild the Docker container; the changes are localized to 'src/dope/inference/cuboid_pnp_solver.py'

Please let me know if this fix helps.

Ademord, 2020-10-07:
@nv-jeff Thanks for taking the time to fix the issue above!
I am testing now the changes you've pushed.

✅ **Inference Test:** works.
**Comment:** So far I really appreciate the output from Dope when its running about X points detected (if you could make comment if you ever try to remove it so that we can manually turn it on).

❌ **Training test:** 3 error encountered so far during training.

🔵 **Error 1:**
```
Traceback (most recent call last):
  File "/root/catkin_ws/src/dope/scripts/train.py", line 52, in <module>
    import ConfigParser
ModuleNotFoundError: No module named 'ConfigParser'
```
**Solution found:** 
ConfigParser is a 2.7 module. 
I added to the ```requirements.txt``` the module ```configparser``` and modified the import at ```train.py``` to do ```import configparser as ConfigParser```.

🔵 **Error 2:** 
```
Traceback (most recent call last):
  File "/root/catkin_ws/src/dope/scripts/train.py", line 1392, in <module>
    _runnetwork(epoch,trainingdata)
  File "/root/catkin_ws/src/dope/scripts/train.py", line 1334, in _runnetwork
    output_belief, output_affinities = net(data)
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/module.py", line 722, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/parallel/data_parallel.py", line 147, in forward
    raise RuntimeError("module must have its parameters and buffers "
RuntimeError: module must have its parameters and buffers on device cuda:1 (device_ids[0]) but found one of them on device: cuda:0
```
**Solution found:** 
When specifying ```--gpu 0 1 2 3 4 5``` GPU 0 must always be mentioned (to reproduce run ```--gpu 1 2 3```)

🔵 **Error 3:** 
```
Traceback (most recent call last):
  File "/root/catkin_ws/src/dope/scripts/train.py", line 1392, in <module>
    _runnetwork(epoch,trainingdata)
  File "/root/catkin_ws/src/dope/scripts/train.py", line 1368, in _runnetwork
    epoch,batch_idx,loss.data[0])
IndexError: invalid index of a 0-dim tensor. Use `tensor.item()` in Python or `tensor.item<T>()` in C++ to convert a 0-dim tensor to a number
```
**Solution found:** 
This is an old error from transitioning from Torch 0.4 to 1.4+.
Solution is to modify lines 1366 to 1381 to:
```
        with open (opt.outf+namefile,'a') as file:
            s = '{}, {},{:.15f}\n'.format(
                epoch,batch_idx,loss.data) 
            # print (s)
            file.write(s)

        if train:
            if batch_idx % opt.loginterval == 0:
                print('Train Epoch: {} [{}/{} ({:.0f}%)]\tLoss: {:.15f}'.format(
                    epoch, batch_idx * len(data), len(loader.dataset),
                    100. * batch_idx / len(loader), loss.data))
        else:
            if batch_idx % opt.loginterval == 0:
                print('Test Epoch: {} [{}/{} ({:.0f}%)]\tLoss: {:.15f}'.format(
                    epoch, batch_idx * len(data), len(loader.dataset),
                    100. * batch_idx / len(loader), loss.data))
```
I have created a pull request to submit the changes for errors 1 and 3 changes. I am currently training and will notify of any future problems.

closed 2020-11-02 by nv-jeff