NVlabs/Deep_Object_Pose #141: Training on Custom Object using NDDS does not detect object after training

sebastian-ruiz, 2020-10-02:
Steps that I took:
- I made a 3d scan of my object using a 3d scanner
- I imported the object into NDDS and generated a training dataset of 100,000 images and a test dataset of 2000 images
- I trained DOPE on this dataset.
- I apply inference to the training data images on DOPE but DOPE does not return any output.

In the loss_train.csv file I see that the loss in epoch 1 is 0.28 and the loss in epoch 60 is 2.9E-09

The training data generated with NDDS looks like this:

[image]
[image]
[image]

I adapted the [script from this issue](https://github.com/NVlabs/Deep_Object_Pose/issues/76) to apply inference to the training images. [This is my script for inference](https://gist.github.com/sebastian-ruiz/41e4d9ca41e5587549f951838e630de3). When using the script on the soup cans and the trained model from the readme, it works perfectly. When applied to my custom object, DOPE does not detect the object class, nor does it detect the pose.

TontonTremblay, 2020-10-02:
Hello, so sorry your model does not work well. There was a similar discussion about a chair detection, https://github.com/NVlabs/Deep_Object_Pose/issues/137. If you could visualize the the belief maps, that might help us understand what the network sees. The loss looks very low, what is the loss at epoch 1-2-3? Could you share a json file from training data.

sebastian-ruiz, 2020-10-12:
I try and get the belief maps using the following code snippet:

```
for j in range(vertex2.size()[0]):
    belief = vertex2[j].clone()
    belief -= float(torch.min(belief).data.cpu().numpy())
    belief /= float(torch.max(belief).data.cpu().numpy())
    belief = torch.clamp(belief, 0, 1)
    belief = torch.cat([belief.unsqueeze(0), belief.unsqueeze(0), belief.unsqueeze(0)]).unsqueeze(0)
    temp = Variable(belief.clone())
    array_belief = temp.data.squeeze().cpu().numpy().transpose(1, 2, 0) * 255
    cv2.imshow('belief_' + str(j), array_belief)
```
where `vertex2` is the variable from the `detect_object_in_image(...)` function in `detector.py` (see my [inference.py script](https://gist.github.com/sebastian-ruiz/41e4d9ca41e5587549f951838e630de3) for more info).

When I try and get the belief maps of the soup cans I get this:

[image]

but when I try the same with my object I get this (this image is from the training set. Using the webcam I also get the same belief maps and no box):

[image]

My `_object_settings.json` file looks like this:

```
{
	"exported_object_classes": [
		"KALO_K1_5_edited"
	],
	"exported_objects": [
		{
			"class": "KALO_K1_5_edited",
			"segmentation_class_id": 144,
			"segmentation_instance_id": 16014600,
			"fixed_model_transform": [
				[ 0, 0, 1, 0 ],
				[ -1, 0, 0, 0 ],
				[ 0, -1, 0, 0 ],
				[ -0.55800002813339233, -0.25029999017715454, 0.22859999537467957, 1 ]
			],
			"cuboid_dimensions": [ 119.63559722900391, 29.607500076293945, 38.399700164794922 ]
		}
	]
}
```
I noticed that my cuboid_dimensions are really in meters, where my object is 119m long even though it should be 11.9cm long. I have therefore set the config_pose.yaml file to the following, so that it corresponds to the size of the object in the training data:
```
dimensions: {
    "KALO_K1_5_edited": [ 119.63559722900391, 29.607500076293945, 38.399700164794922 ],
   ...
}
mesh_scales: {
    "KALO_K1_5_edited":  1.0,
   ...
}
```

The losses at the end of each epoch 1, 2, 3, 4, 5 from `loss_train.csv` are:
```
1, 624,0.000003241889772
2, 624,0.000002674767757
3, 624,0.000000930202077
4, 624,0.000000722245716
5, 624,0.000001037328047
...
15, 624,0.000000092100692
...
30, 624,0.000000032321594
```

TontonTremblay, 2020-10-12:
Your code to display the belief maps is not correct. DOPE has two outputs, you are using the vertex map, the one that is pointing toward the center of the object. 

Here is my code I am currently using (sorry I need to update this repo): 
```

    @staticmethod
    def detect_object_in_image(net_model, pnp_solver, in_img, config, 
            grid_belief_debug = False, norm_belief=True,run_sampling=False,network='dope'):
        ''' Detect objects in a image using a specific trained network model
            Returns the poses of the objects and the belief maps
            '''
        if in_img is None:
            return []

        if network == 'full':
            scale_factor = 1
            OFFSET_DUE_TO_UPSAMPLING = 0 
        else: # 'dope' and 'mobile'
            scale_factor = 8
            # OFFSET_DUE_TO_UPSAMPLING = 0.4395
            OFFSET_DUE_TO_UPSAMPLING = 0

        # print("detect_object_in_image - image shape: {}".format(in_img.shape))

        # Run network inference
        # print(in_img.shape)
        image_tensor = transform(in_img)
        image_torch = Variable(image_tensor).cuda().unsqueeze(0)
        # print(image_torch.shape)
        out, seg = net_model(image_torch)  # run inference using the network (calls 'forward' method)
        vertex2 = out[-1][0]
        aff = seg[-1][0]

        # Find objects from network output
        detected_objects = ObjectDetector.find_object_poses(vertex2, aff, pnp_solver, config,
            run_sampling=run_sampling,
            scale_factor = scale_factor,
            OFFSET_DUE_TO_UPSAMPLING = OFFSET_DUE_TO_UPSAMPLING)

        if not grid_belief_debug: 

            return detected_objects, None
        else:
            # Run the belief maps debug display on the beliefmaps
            
            upsampling = nn.UpsamplingNearest2d(scale_factor=scale_factor)
            tensor = vertex2
            belief_imgs = []
            in_img = (torch.tensor(in_img).float()/255.0)
            in_img *= 0.7            

            for j in range(tensor.size()[0]):
                belief = tensor[j].clone()
                if norm_belief:
                    belief -= float(torch.min(belief).item())
                    belief /= float(torch.max(belief).item())

                # print (image_torch.size())
                # raise()    
                # belief *= 0.5
                # print(in_img.size())
                belief = upsampling(belief.unsqueeze(0).unsqueeze(0)).squeeze().squeeze().data 
                belief = torch.clamp(belief,0,1).cpu()  
                belief = torch.cat([
                            # belief.unsqueeze(0) + in_img[:,:,0],
                            # belief.unsqueeze(0) + in_img[:,:,1],
                            # belief.unsqueeze(0) + in_img[:,:,2]
                            belief.unsqueeze(0),
                            belief.unsqueeze(0),
                            belief.unsqueeze(0)
[... 94 lines cut ...]
                    .copy_(tensor[k])
                k = k + 1
        return grid

    @staticmethod
    def get_image_grid(tensor, filename, nrow=3, padding=2,mean=None, std=None):
        """
        Saves a given Tensor into an image file.
        If given a mini-batch tensor, will save the tensor as a grid of images.
        """
        from PIL import Image
        
        # tensor = tensor.cpu()
        grid = ObjectDetector.make_grid(tensor, nrow=nrow, padding=10,pad_value=1)
        if not mean is None:
            # ndarr = grid.mul(std).add(mean).mul(255).byte().transpose(0,2).transpose(0,1).numpy()
            ndarr = grid.mul(std).add(mean).mul(255).byte().transpose(0,2).transpose(0,1).numpy()
        else:      
            ndarr = grid.mul(0.5).add(0.5).mul(255).byte().transpose(0,2).transpose(0,1).numpy()
        im = Image.fromarray(ndarr)
        # im.save(filename)
        return im
```

try to run this please and report what you get.

sebastian-ruiz, 2020-10-13:
Thank you for the code. Here are the results:

Verifying that it works correctly with the soup can:

[image]

Trying on image from test set:

[image]

Trying on webcam:

[image]

From this I would say that something has gone wrong with the training.

TontonTremblay, 2020-10-13:
Could you visualize the gt belief maps you generate when training. One of these,  https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/train.py#L1338 . I think there is something going on with your training data.

TontonTremblay, 2020-10-13:
```
	"exported_objects": [
		{
			"class": "KALO_K1_5_edited",
			"segmentation_class_id": 144,
			"segmentation_instance_id": 16014600,
			"fixed_model_transform": [
				[ 0, 0, 1, 0 ],
				[ -1, 0, 0, 0 ],
				[ 0, -1, 0, 0 ],
				[ -0.55800002813339233, -0.25029999017715454, 0.22859999537467957, 1 ]
			],
			"cuboid_dimensions": [ 119.63559722900391, 29.607500076293945, 38.399700164794922 ]
		}
```
Where are the projected cuboid points?

thangt, 2020-10-13:
@TontonTremblay That is the object_settings.json file which describe the object to capture. The projected_cuboid is in each of the captured frame data: 000001.json ...

sebastian-ruiz, 2020-10-14:
Could it be that there is something wrong with my training data? Here is the data for 000001. In unreal engine the `segmentation id assignment type` and `segmentation id assignment type` are both set to `spread evenly`.

000001.cs.png:
[image]

000001.depth.png:
[image]

000001.is.png:
[image]
000001.png:
[image]

And the [000001.json](https://gist.github.com/sebastian-ruiz/8c91b8b08e4769bfebddeeb0045c000e) file. These 4 images and 1 json file make up the training data for this item.

sebastian-ruiz, 2020-10-14:
I think I am making a mistake in `config_pose.yaml`. Namely, in class_ids I do the following:

```
class_ids: {
    "cracker": 1,
    "gelatin": 2,
....... [omitted] ......
    "GreenBeans"        : 35, 
    "PeasAndCarrots"    : 36,
    "KALO_K1_5_edited"  : 37 
}
```
I think that the id of "KALO_K1_5_edited" shoud be that picked by NDDS in `*****.cs.png`.

TontonTremblay, 2020-10-14:
Class id should not impact the algorithm. Can you try to render the belief map from a GT you generated please? See above which line in the training.

sebastian-ruiz, 2020-10-21:
Sorry for the delay. I am retraining the model and storing the `target_belief` (https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/train.py#L1338) for every epoch. I will report back.

Gaoee, 2021-01-14:
Hi, I am also trying to use NDDS to generate a new data set. But how can I  import my FBX format 3D model and replace the original object in NDDS. 
Thank you.

closed 2025-08-15 by nv-jeff