NVlabs/Deep_Object_Pose #145: Training on new Object

Ixion46, 2020-10-20:
Hey Guys, thanks for sharing your great work.
I want to test that on my own dataset and training data that I created with NDDS. Which Data do I need to export for the training data? The color images and the instance segmentation? Is it important which value the object has in the image of the instance segmentation when I read the image as np.array(000000.is.png)?
I were not able to find answers in the code for my questions.
Best Regards Ixion

TontonTremblay, 2020-10-20:
You only need the rgb images. DOPE does not use the segmentation, sorry for the confusion.

Ixion46, 2020-10-21:
Thanks for your answer, that helps a lot. I usually translate my models to origin and then axis align them, but with one model I do not get accurate results if i plot the pointcloud of the model with the NDDS gt. 
So my question is now when I use this not axis-aligned model , will DOPE give me the bounding box showed on the image (because these are the values of projected cuboid I guess) or the "best" bounding box along the object? And do I need to train with single objects on the images or can I put like multiple different objects in every scene? 
I hope you can help me again
[image]

TontonTremblay, 2020-10-21:
DOPE is agnostic of where the bounding box is, so if you exported this 3d model with NDDS and trained DOPE with that, it should work fine. Just make sure the cuboid dimensions are the ones that have the crooked 3d models within.

Ixion46, 2020-10-21:
Thanks that helps a lot, but do you also have an answer for the 2nd part of the question?
Will it work if I train with 6 annotated objects in any scene?
[image]

TontonTremblay, 2020-10-21:
I see some some objects with symmetries, e.g., cubes and plate. Dealing with symmetries has not been really explored with DOPE. I had some ideas, but they are not straight forwards to implement and test.

Ixion46, 2020-10-21:
I have planned to show only 3 sides of the cubes and to rotate the plate only in one axis on the training images, maybe it will work. I will test it out and let you know :).

TontonTremblay, 2020-10-21:
Let me know how it goes!

Ixion46, 2020-10-22:
The bottle and cup work great so far! I have training data of 40k images, with all 6 objects on all images.
When I trained the bottle and cup , I had 4000 batches with a batchsize of 20. Now when I train on the green cube, I only have 2000 batches with a batchsize of 20. Why does it differ? I only have one folder with the images

Ixion46, 2020-10-23:
Somehow the loss of the green cube stays at around 0.002 the whole time, but when I try to detect a cube, it does not detect the cube. Is it because I only show the cube from some specific side or will it simply not work nevermind which training images I will create?

TontonTremblay, 2020-10-23:
The cube is hard because of symmetries... I would debug the belief map of the inference to see what is going on. I shared the code I use in an other thread.

Ixion46, 2020-10-23:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Ixion46, 2020-10-23:
The belief maps does look good for me. Maybe I did something else wrong

TontonTremblay, 2020-10-23:
This looks good to me, I am not sure what is going on, in the config file can you decrease 
`thresh_map: 0.005`
`thresh_points: 0.05`
to the values above and try again?

Ixion46, 2020-10-23:
Thanks for your fast replies! Still no detection, the belief maps are close to the others.

Ixion46, 2020-10-23:
[image]
I get a detection if I use thresh_map = 0.001 and thresh_points = 0.01

TontonTremblay, 2020-10-23:
tweaking is important :P This looks very good, how does it look in the real world. 

The first term is used after smoothing has been applied to the belief map. This is to remove local noise. Then if the threshold you find is too low compared to the neighbors, then it is ignored. If you get that value too low you might get a lot of false positive, be careful when playing around on real data. The second term is about the height peak, e.g., the absolute value, it is somewhat related to the previous one. Anyway, not sure why that object behaves like this.

Ixion46, 2020-10-23:
I have no opportunity to test it on real data at the moment. Maybe in 2 weeks, then I will definitely report to you. Still do not always get a detection, when I checked on 4-5 other images right now.

TontonTremblay, 2020-10-23:
So in order to display them I normalize the belief maps, which makes the peak  really punchy. But if you remove that, I am sure the peak are quite soft... 

I am sure you can find images of cubes online to try it out.

Ixion46, 2020-10-23:
I think I went too far :D
[image]

I have 2 questions in my mind. Is it okay for you , if I try to combine your work with Densefusion? Amazingly I have good results on the cubes with Densefusion, but not with the other objects and with your work the other way round.

And have you tested a multiple camera setup on dope?

TontonTremblay, 2020-10-23:
Can you send me a message to jtremblay@nvidia.com ;)

Ixion46, 2020-10-24:
After a lucky training epoch a got a loss of 0.0015 for the cube and got a detection now in all tested cases, even without adjusting the tresh_map and thresh_points. :)

closed 2020-10-29 by Ixion46

YeeCY, 2020-12-07:
@Ixion46 Could you please explain how you import the red and green boxes into the scene?

Ixion46, 2020-12-07:
Ofcourse. I use the output of the projected points given by the network and project them into the images.

YeeCY, 2020-12-09:
> Ofcourse. I use the output of the projected points given by the network and project them into the images.

Oh, sorry for my unclear question. Actually, I want to ask how did you import the red and green boxes into UE4 (or you just create this object in UE4) to create the training dataset?

Ixion46, 2020-12-09:
I think you mean the orange and yellow boxes? These are the predicted outputs from DOPE and are not in UE4

YeeCY, 2020-12-09:
> I think you mean the orange and yellow boxes? These are the predicted outputs from DOPE and are not in UE4

Nope. I mean the green and red cubes in the image, which is the input to the algorithm, not the yellow and orange bounding boxes predicted by DOPE. I just wonder how did you create the training dataset.

Ixion46, 2020-12-09:
Ah now I get it. These are objects in our university lab. I scanned these objects with an artec 3D scanner, edited the models with meshlab and then converted the resulting models to .obj and imported them into UE4.

YeeCY, 2020-12-10:
> Ah now I get it. These are objects in our university lab. I scanned these objects with an artec 3D scanner, edited the models with meshlab and then converted the resulting models to .obj and imported them into UE4.

Okay, I get it.

blaine141, 2021-01-14:
How big was the dataset that you used when training this and for how many epochs did you train? I am not having much luck training.