NVlabs/Deep_Object_Pose #148: Can DOPE architecture, in theory, learn distances too?

AOL283, 2020-11-11:
Hi!
I am just interested to know whether such a network is able to learn distances (un-normalized vectors), my initial results points to a no though.
Would be glad for any comment!

TontonTremblay, 2020-11-11:
I never tried it, I am sure you could eventually succeed with a lot of changes and loss changes and probably more. Since the objects are known I never investigated that. I am sorry you did not succeed, maybe if you tell me more about your context I could help you out more?

AOL283, 2020-11-12:
Hi!
Thank you for your reply.
Yes, a careful consideration of the loss function is of a big importance, especially if you switch to lower radius ellipses or possibly even points.
I am trying to use DOPE for the detection, 2D model generation of piece-wise linear objects.
Thank you for your clarifications, I'll post an update if I achieve interesting results!

TontonTremblay, 2020-11-12:
so the radius for ellipses was picked to assure best sub pixel prediction. Since you predict in reduced scale by 8, you need to make sure you are not going to see huge pixel shift in the original frame. For example, input is 400x400 and the output is 50x50. When finding a point I do a local [weighted average](https://github.com/NVlabs/Deep_Object_Pose/blob/master/src/dope/inference/detector.py#L323-L349). If you predict a single pixel like you are suggesting you might be getting larger error than you want.

An other approach is to change the output to something that does not change the input size. I have done something like this with 2stage hourglass architecture. But even there you want to produce subpixel results to allow for fine prediction. The performance of that algorithm was also better,30-40% on some objects. 

What do you mean by piece-wise linear objects?

closed 2025-08-15 by nv-jeff