NVlabs/Deep_Object_Pose #150: ModuleNotFoundError 'camera_info_manager' when running in docker

HUSH-Wang, 2020-11-14:
I was running DOPE in a Docker Container.
When running cmd `roslaunch dope camera.launch` in step 2 Start camera node.
There is an error said ModuleNotFoundError: No module named 'camera_info_manager'.
Is there anything I missed?

[image]

TontonTremblay, 2020-11-14:
@nv-jeff do you have an idea?

mintar, 2020-11-16:
When the Dockerfile was upgraded to ROS Noetic, the "rosdep" lines were removed:

https://github.com/NVlabs/Deep_Object_Pose/commit/6b78c466b3d739d3f6cc27c4680df179a3c8f53d#diff-fd78dacf04884694bb696c4f604cc27de9f3372c43531517a3205bcbcc28eee1L60-L63

Running rosdep takes care of automatically installing all dependencies that are specified in the `package.xml`. So either you put these lines back in (that would be the clean preferred version), or you manually install the package `ros-noetic-camera-info-manager-py`.

HUSH-Wang, 2020-11-17:
Thank you so much. It works.

closed 2020-11-17 by HUSH-Wang