NVlabs/Deep_Object_Pose #151: failed detection of the custom object

mbajlo, 2020-11-21:
Hello all,
I have read a lot of topics here, it got me pretty far, but now I am stuck.
I am trying to make train the network on my custom object which is made using the blender, exported and imported in the NDDS as fbx file. In the NDDS I have made the train dataset of 200 images and the test dataset of 50 images. 
the  `_camera_settings.json` looks like this:
```
{
	"camera_settings": [
		{
			"name": "Viewpoint",
			"horizontal_fov": 90,
			"intrinsic_settings":
			{
				"resX": 400,
				"resY": 400,
				"fx": 200,
				"fy": 200,
				"cx": 200,
				"cy": 200,
				"s": 0
			},
			"captured_image_size":
			{
				"width": 400,
				"height": 400
			}
		}
	]
}
```

the `_object_settings` looks like this:
```
{
	"exported_object_classes": [
		"thorHammer_387"
	],
	"exported_objects": [
		{
			"class": "thorHammer_387",
			"segmentation_class_id": 255,
			"segmentation_instance_id": 16777214,
			"fixed_model_transform": [
				[ 0, 0, 1, 0 ],
				[ -1, 0, 0, 0 ],
				[ 0, -1, 0, 0 ],
				[ 0, -100, -420, 1 ]
			],
			"cuboid_dimensions": [ 8.1920995712280273, 38.728199005126953, 13.551199913024902 ]
		}
	]
}
```

the `000019.json`(the other ones are quite similar) for example looks like this:
```
{
	"camera_data":
	{
		"location_worldframe": [ -405.77529907226563, 9.6858997344970703, 100 ],
		"quaternion_xyzw_worldframe": [ 0, 0, -0.95569998025894165, 0.29449999332427979 ]
	},
	"objects": [
		{
			"class": "thorHammer_387",
			"instance_id": 16777214,
			"visibility": 1,
			"location": [ 0, 0, 17.209299087524414 ],
			"quaternion_xyzw": [ 0, 0.95569998025894165, 0, 0.29449999332427979 ],
			"pose_transform": [
				[ 0.56279999017715454, 0, -0.82660001516342163, 0 ],
[... 196 lines cut ...]
    with open("C:\\Users\\m\\Desktop\\debug\\target_belief_20.pt", 'rb') as f:
        buffer = io.BytesIO(f.read())
    LodedTensor = torch.load(buffer, map_location='cpu')

    # show the target belief for each batch
    count = 0
    for t in LodedTensor:
        print ("tensor_{}, {}".format(count, t.shape))
        count += 1
        im_belief = targetBelif_show(t, norm_belief=True)
        cv_imageBelief = np.array(im_belief)
        imageToShow = cv2.resize(cv_imageBelief, dsize=(800, 800))
        cv2.imshow('beliefMaps', imageToShow)
        cv2.waitKey(0)    

main()
```

But then I have seen that the targetBeliefMaps are all the same no matter which epoch tensor I try to read and I would say these are not looking good.

[image]

I know I am missing something big and I am not sure if I am showing the target belief maps from training correctly, can you help or point me in the right direction?

Thanks, any help is appreciated.

TontonTremblay, 2020-11-22:
Thank you for the detailed information. It sounds like things should work out of the box, but they are not. I would say that the example with hammer you share, the hammer is too close to the camera, you can only see a 2 keypoints. Is your data all looking like this one. I have noticed that if the keypoints are not all seen by the image, the architecture is hard to train to produce valid detection. Maybe try with a smaller hammer appearance . 

It looks like the training data is empty as well, I am not sure why. You could check in the dataloader if the belief maps are correct at that level.

mbajlo, 2020-11-23:
Hello @TontonTremblay , thank for the reply
The image I have showed is just one of the many in the sequence. For example this is how the image 000120 looks like:
[image]
I am changing the distance and the angle.

Also I have tried to use the option `Auto Start Capturing` in the `SceneCapturer_Simple` and then it start from the center of the object's coordinate system. I have also tried to create a dataset without the `Auto Start Capturing` option, then I have to start capturing by myself but the situation is the same.
I am using the `OrbitalMovement` to change the Yaw angle and I was thinking to use the network trained on the Yaw angle to retrain later with the Pitch angle, because I couldn't find a way to change both angles at the same time. If I want to change both values in the NDDS it looks like the Yaw angle has a priority.

When you say 2 keypoints, what do you mean?

In the video `data_overview.mp4` it says (around 20sec) if the visibility tag is `0` is fully visible and if it is `1` then it is fully invisible, on all my `.json` files I have value `1`(fully invisible), how can I change it to be `0`? Could this cause the empty training data?

I tried to debug the `train.py` to check `training data`, but I can see the path given in the `--data` variable is used and it recognizes the `images `and belonging `.json`
[image]

Do you see something wrong here?
It seems it populated the trainingdata just fine for me. The same is worth for the `testdata`

This is how the output looks like when I start training:
```
start: 08:34:17.726571
load data
training data: 4 batches
testing data: 1 batches
```

So this line printed the line [training data: 4 batches](https://github.com/NVlabs/Deep_Object_Pose/blob/11bbc3b8545e099b35901a13f549ddddacd7dd1f/scripts/train.py#L1300) 
which means the trainingdata is not `none`at least.

Is there an explanation how to show `belief maps` at the `dataloader `level? Could you help me with that?

Thanks

TontonTremblay, 2020-11-24:
Something wrong is going on, I am not sure what. How do you call the training script?

mbajlo, 2020-11-24:
hello @TontonTremblay ,

Thanks for the reply, the training script name is the same, I kept it as `train.py`. Do you think the name could make a difference?

TontonTremblay, 2020-11-24:
How do you call it. Like python train.py--data.... ? (Sorry phone answer)

mbajlo, 2020-11-24:
Hello @TontonTremblay 

You meant how do I execute the training script, I do it like this from the folder where the `train.py` is:

`python train.py --data C:\Users\m\Desktop\ML\ThorHammer_387mm_dataset\train\yaw\thorHammer_YawAngle\ --batchsize 50 --pretrained False --datatest C:\Users\m\Desktop\ML\ThorHammer_387mm_dataset\test\yaw\thorHammer_YawAngle\ --imagesize 400 --object thorHammer_387 --workers 8 --outf markoThorHammer_testBeliefExport`

Or if do the debug in the VS code I just execute the launch configuration with the same `args` given before, but it is basically the same thing.
`

TontonTremblay, 2020-11-24:
In your dataset do you only have the hammer? If so remove the --object in
the python.

mbajlo, 2020-11-24:
Yes that's right, I have only the hammer. I will remove the `--object` and try again.
I will let you know what happened.

Thanks

mbajlo, 2020-11-27:
Hello @TontonTremblay 

I have retrained the network again, the `target belief maps` are now looking like this:

[image]

So I would say it looks pretty good.

But (there is always a but :D ) after I try to test my network on the images from the `train` or `test` datasets or if I make a new image of the hammer, the `results list` is always empty and the `belief maps` are looking something like this:

[image]

I tried to train the network with more than one object in the image, or by changing the yaw angle or the pitch angle. The `target belief maps` are looking always getting back as in the first image, but I cannot detect an object on the image from the `train` or `test` dataset and the `belief maps` are looking strange like above.

Do you know what could be wrong, what should be my next step?

TontonTremblay, 2020-11-27:
On the phone again.

I think you need to train for longer. I normally train for 60 epochs on a
dataset of 120k images and with a batch size of 128 on 4 p100.

mbajlo, 2020-12-02:
Hello @TontonTremblay 

I thought you are using the NVidia GPUs, but I have the a Radeon GPU so I can train only with my CPU. 
Just for testing I have made a training dataset of 100 images of the hammer (and the test dataset of 20 images), all images have been taken from the same position of the camera (there are no transport, no movement and not angle change), when I train the network for 60 epochs I get this response when I test the network:

list of found objects: []

[image]

[image]

Do you think the problem could be in `visibility` , should it be 0 for full visibility?
The image size is `400x400`, should I make it `800x800` or more?

```
{
	"camera_data":
	{
		"location_worldframe": [ -990, -80, 21.502099990844727 ],
		"quaternion_xyzw_worldframe": [ 0, 0, 0, 1 ]
	},
	"objects": [
		{
			"class": "thorHammer_387",
			"instance_id": 16777200,
			"visibility": 1,
			"location": [ 0, -4.4168000221252441, 49.195098876953125 ],
			"quaternion_xyzw": [ 0, -0.70709997415542603, 0, 0.70709997415542603 ],
			"pose_transform": [
				[ -1, 0, 0, 0 ],
				[ 0, 0, 1, 0 ],
				[ 0, -1, 0, 0 ],
				[ 0, -4.4168000221252441, 49.195098876953125, 1 ]
			],
			"cuboid_centroid": [ -0.087600000202655792, 4.3204998970031738, 48.991600036621094 ],
			"projected_cuboid_centroid": [ 199.64230346679688, 217.63749694824219 ],
			"bounding_box":
			{
				"top_left": [ 134.22529602050781, 169.74690246582031 ],
				"bottom_right": [ 296.0574951171875, 229.5238037109375 ]
			},
			"cuboid": [
				[ -6.8632998466491699, -15.043600082397461, 53.087600708007813 ],
				[ -6.8632001876831055, -15.043600082397461, 44.895599365234375 ],
				[ -6.8632001876831055, 23.684499740600586, 44.895599365234375 ],
				[ -6.8632998466491699, 23.684499740600586, 53.087600708007813 ],
				[ 6.6880002021789551, -15.043600082397461, 53.087699890136719 ],
				[ 6.6880002021789551, -15.043600082397461, 44.895599365234375 ],
				[ 6.6880002021789551, 23.684499740600586, 44.895599365234375 ],
				[ 6.6880002021789551, 23.684499740600586, 53.087699890136719 ]
			],
			"projected_cuboid": [
				[ 174.14370727539063, 143.32530212402344 ],
				[ 169.42579650878906, 132.98399353027344 ],
				[ 169.42579650878906, 305.50930786132813 ],
				[ 174.14370727539063, 289.22799682617188 ],
				[ 225.19599914550781, 143.32539367675781 ],
				[ 229.79359436035156, 132.98399353027344 ],
				[ 229.79359436035156, 305.50930786132813 ],
				[ 225.19599914550781, 289.2279052734375 ]
			]
		}
	]
}
```
Or should I just make a dataset with more than 1000 images and then train the network for 60 epochs?

Could it be the losses of the epoch 60 are still too high?
`test loss 60, 0,0.007053715176880`  
`train loss 60, 0,0.007159050088376`

How low the loss should be?

TontonTremblay, 2020-12-07:
Well I am not sure what to day here, you need a fairly large dataset 5k to 60k and you need to train for 2-3 days. I am sorry I am not sure CPU train will work...

closed 2025-08-15 by nv-jeff