NVlabs/Deep_Object_Pose #153: Unable to train a working model

blaine141, 2020-12-24:
Hello, 

I tried training on a custom data set earlier, and it failed to detect the object in Gazebo. Instead, I backed down to training the soup in the "fat" dataset. That also failed to detect anything. Finally, I tried training the soup on just the soup folder of the fat dataset. That also failed to detect anything. 

I can use the pre-trained weights to detect the soup can in Gazebo, so it seems I do not understand how to train this repo. Below are the commands I used. Is there anything that I am missing?

First, I tried just running the command `python train.py --data /mnt/Data/fat --object soup --batchsize 16 --gpuids 0 --outf soup --epochs 15`. This ran, the loss fluctuated, but I think that was just because the dataset included batches that did not contain the soup. 

Next I tried matching the class name and only using the single soup directory. `python train.py --data /mnt/Data/fat/single/005_tomato_soup_can_16k --object 005_tomato_soup_can_16k --batchsize 16 --gpuids 0 --outf 005_soup --epochs 15`. This also yielded the same results. The loss just immediately went to and hovered around 0.009

So, I have the following questions about training.
1. What does the --object parameter need to be. Does it need to match the class? I saw in train.py your example just uses soup but I do not see where that string is used anywhere else or defined to equal 005_tomato_soup_can_16k
2. How are objects indexed and associated with the string names. Where is the indexing defined, and how do I set that up for custom objects. For any of my fat .pth files, will I need to change the indexing in config_pose.yaml
3. If I want to detect multiple objects, should I make multiple .pth files or can I include them all in one training? How do I define that
4. How many epochs should I run and what is the expected loss? I've seen mixed recommendations in the issues.
5. Any additional recommendations running on one RTX 2070 besides reduce the batch size to 16 for memory?
6. I notice there are depth images in the fat data set. DOPE does not use depth, correct?

Sorry for so many questions. Thank you for your time

saratrajput, 2020-12-24:
I'll try to answer some of your questions. First, you can take a look at the [wiki](https://github.com/NVlabs/Deep_Object_Pose/wiki) to understand how to train the model for a custom object.

1. The --object parameter indicates the name of the mesh (or the tag, if you've changed it) when you create your dataset following the wiki instructions.
3. If you want to detect multiple objects in the same scene, you need to train multiple models. Or you could train a single model with multiple objects in it, but in this case you won't be able to distinguish between the objects. I'm assuming you want the former scenario.
4. [Wiki](https://github.com/NVlabs/Deep_Object_Pose/wiki) mentions 30 epochs.
6. No, DOPE doesn't use depth images.

Sorry question 2 wasn't clear for me. And for question 5, I don't have any additional recommendations.

blaine141, 2020-12-24:
Thank you so much. I completely missed that there was a wiki. Explains why I was so lost. Your explanation of how the models detect one object make sense. Not quite sure why I was confused. Thank you very much.

closed 2020-12-24 by blaine141

blaine141, 2020-12-24:
Want to share that I looked in the code. The way --object works is that if that string is in the class name, it selects the class. so soup will select tomato-soup. --object must be all lowercase