NVlabs/Deep_Object_Pose #157: continue with a trained model

fehimkus, 2021-01-19:
hello,

I wanted to ask 2 questions,

1- How can I continue training with a ready model, for example I have net_epoch_84, and I want to load this model. In python code I found this line to load but I can not be sure.

original
1308   if opt.net != '':
1309       net.load_state_dict(torch.load(opt.net))

changed
1308   if opt.net != '':
1309       pathh = "/content/drive/MyDrive/pthlar/net_epoch_84.pth"
1310       net.load_state_dict(torch.load(pathh))

2- I am exporting such a file from NDDS that contain only png type and you can see here. I did not use scene and approximately 5000 capture I stocked. Do you think it is proper way to train? I would be appreciated any help.

thank you

[image]
[image]
[image]

TontonTremblay, 2021-01-20:
This look good to me.

fehimkus, 2021-01-24:
Why in this picture the object is green? I am using GroupActorManager_BP