NVlabs/Deep_Object_Pose #160: solvePnP error for less than 6 valid points

geotsam, 2021-01-28:
Hi! I am using DOPE for research purposes for detecting objects for manipulation and I have a problem. In case the valid points found are 6 or more the detection works as expected. Otherwise an opencv error is thrown:

```
(6, 'valid points found')
(6, 'valid points found')
(5, 'valid points found')
OpenCV(3.4.1) Error: Assertion failed ((((_src) != __null && (((const CvMat*)(_src))->type & 0xFFFF0000) == 0x42420000 && ((const CvMat*)(_src))->cols > 0 && ((const CvMat*)(_src))->rows > 0) && ((const CvMat*)(_src))->data.ptr != __null) && (((_dst) != __null && (((const CvMat*)(_dst))->type & 0xFFFF0000) == 0x42420000 && ((const CvMat*)(_dst))->cols > 0 && ((const CvMat*)(_dst))->rows > 0) && ((const CvMat*)(_dst))->data.ptr != __null) && (_src->rows == 1 || _src->cols == 1) && (_dst->rows == 1 || _dst->cols == 1) && _src->cols + _src->rows - 1 == _dst->rows + _dst->cols - 1 && (((_src->type) & ((1 << 3)*512 - 1)) == (((5) & ((1 << 3) - 1)) + (((2)-1) << 3)) || ((_src->type) & ((1 << 3)*512 - 1)) == (((6) & ((1 << 3) - 1)) + (((2)-1) << 3))) && (((_dst->type) & ((1 << 3)*512 - 1)) == (((5) & ((1 << 3) - 1)) + (((2)-1) << 3)) || ((_dst->type) & ((1 << 3)*512 - 1)) == (((6) & ((1 << 3) - 1)) + (((2)-1) << 3)))) in cvUndistortPointsInternal, file /io/opencv/modules/imgproc/src/undistort.cpp, line 384
cv2.solvePnP failed with an error
(6, 'valid points found')
(6, 'valid points found')
```

Since I haven't seen any similar issue, I'm guessing there's something wrong from my side, but I cannot seem to find what. Do you have any idea why this is failing?

pablodot, 2021-02-06:
Hi!
take a look at this:
https://github.com/NVlabs/Deep_Object_Pose/commit/2df0c0b0eda8344e1fd752f4f84c2a83e512bea5

geotsam, 2021-02-06:
I've seen this and I'm on the latest commit. It's exactly there that the error is raised when using `pnp_algorithm = cv2.SOLVEPNP_P3P`.

pablodot, 2021-02-08:
OK. I had a similar problem and debugged it printing the exception, replacing lines 106 to 109 with "except Exception as e: print(e)".
My cv2 exception was about lack of camera calibration, because I didn't use default camera script and mine wasn't calibrated.

Print your exception and probably you will find out next step