NVlabs/Deep_Object_Pose #167: dope: Cannot locate rosdep definition for [camera_info_manager_py]

ABHIINAV12, 2021-04-08:
I am following up the instructions on the github page.
I have installed everything in the latest version just a day before.
I am facing an issue with running a rosdep update command "rosdep install --from-paths src -i --rosdistro noetic".
Which is giving me the error : "dope: Cannot locate rosdep definition for [camera_info_manager_py]".
After googling the error for a while, I changed the command a bit to "rosdep install --from-paths  --ignore-src --rosdistro neotic -y".
Which is now giving me the error of "dope: Cannot locate rosdep definition for [vision_msgs]".

nv-jeff, 2021-04-08:
Running DOPE does not require the 'camera_info_manager_py' package; this is an error in the package.xml file that we will fix.

However, like you I am unable to install this package using 'rosdep install' and I'm not sure why. Perhaps because camera_info_manager_py has not been officially upgraded to python3? 

At any rate, you can manually install it with the following steps:

```
cd ~
git clone https://github.com/ros-perception/camera_info_manager_py
cp -r camera_info_manager_py/src/camera_info_manager /opt/ros/noetic/lib/python3/dist-packages/
```
I tested this within the Docker container and it works.

nv-jeff, 2021-04-08:
Running DOPE itself does not require the 'camera_info_manager_py' package, although our 'camera' node does use it.  Like you, I am unable to install this package using 'rosdep install' and I'm not sure why. Perhaps because camera_info_manager has not been officially upgraded to python3? 

At any rate, you can manually install it with the following steps:
```
cd ~
git clone https://github.com/ros-perception/camera_info_manager_py
cp -r camera_info_manager_py/src/camera_info_manager /opt/ros/noetic/lib/python3/dist-packages/
```

I tested this (within the Docker container) and it works.

haijunsu-osu, 2021-04-26:
I tried

> Running DOPE itself does not require the 'camera_info_manager_py' package, although our 'camera' node does use it. Like you, I am unable to install this package using 'rosdep install' and I'm not [...]

I tried this method. It did not work on my system. The folder "camera_info_manager" was indeed copied to " /opt/ros/noetic/lib/python3/dist-packages/". 

However I still got the error ""dope: Cannot locate rosdep definition for [camera_info_manager_py]"" Any clue?

nv-jeff, 2021-04-26:
Ah, I what I meant was that by manually installing it, you no longer need to run rosdep to install it. If you're using rosdep for other purposes, you may edit 'package and remove (or comment out) the line:
`  <depend>camera_info_manager_py</depend>
`

mintar, 2021-04-28:
> Like you, I am unable to install this package using 'rosdep install' and I'm not sure why. Perhaps because camera_info_manager has not been officially upgraded to python3?

Yes, exactly. The database of rosdep keys for Noetic is here:

https://github.com/ros/rosdistro/blob/master/noetic/distribution.yaml

The one for Melodic is here:

https://github.com/ros/rosdistro/blob/master/melodic/distribution.yaml

You can see that there's a key for `camera_info_manager_py` in Melodic, but not Noetic. This means that the package has not been "released" into Noetic yet.

The steps to get this fixed (if someone's interested) are:

* clone https://github.com/ros-perception/camera_info_manager_py into your catkin workspace
* convert it to Python3. I just checked, and it looks like the code is already supposed to work on both Python2 and Python3, so maybe there's nothing to do here.
* Check that `rosdep install --from-paths . -i --rosdistro noetic` works. You'll probably have to update some keys in the `package.xml` file (`python-rospkg` -> `python3-rospkg`, `python-yaml` -> `python3-yaml`). See https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml for the full list of available python keys.
* Test that everything works (e.g., by running the `camera` node from DOPE) on Noetic
* Submit a pull request
* The maintainer of `camera_info_manager_py` has to do a "bloom release". This will insert the rosdep key into the files above and fix this error for everyone.

BTW, the proper way to fix problems with missing binary packages is to clone the missing package into your catkin workspace instead of messing with `/opt`. That way, rosdep will recognize the package, and the error will disappear. This is what `rosdep install -i` does (`-i` means `--ignore-packages-from-source`, so if the source package is found in the catkin workspace, rosdep won't try to install it from binaries).

andrei91ro, 2021-11-27:
I also had this problem when trying to start the webcam publishing node using `roslaunch dope camera.launch`.

My solution was to clone the camera_info_manager_py package and install it with pip3:
```
git clone https://github.com/ros-perception/camera_info_manager_py
cd camera_info_manager_py
pip3 install ./
```

Should pip3 complain about not finding `yaml` package, you can simply remove it by editing the file `camera_info_manager_py/setup.py` so the install requirements look like:
```
    install_requires=['rospkg'],
````
After installing the package with pip3 the camera launcher will work fine.