NVlabs/Deep_Object_Pose #170: Failed: no cuda capable devices is found

sivaeinfo, 2021-05-18:
Actually i have tried to use the dope package in https://github.com/yehengchen/DOPE-ROS-D435

I have installed require packages and dependencies.
In this, they have given two command.
1.roslaunch realsense2_camera rs_rgbd.launch
2. roslaunch dope dope.launch

This gives following error
[image][image]

Please help me to sort out this issues.

TontonTremblay, 2021-05-18:
do you have an NVIDIA's GPU?

sivaeinfo, 2021-05-19:
I have NVIDIA GT218 GeForce 210

nv-jeff, 2021-09-27:
The GeForce 210 is too old to run DOPE. The last supported CUDA version for this card is CUDA 6.5 so you will not be able to upgrade to 9.2.

The Jetson NANO will run DOPE, and you should be able to install it using the directions in the main README.

sivaeinfo, 2021-09-28:
Now I had GEFORCE GTX 1050 ti, Is ut enough to run DOPE.

nv-jeff, 2021-09-28:
Yes, the 1050TI should work.