NVlabs/Deep_Object_Pose #172: TypeError : Caught TypeError in DataLoader worker process 0

Chi-Ruei, 2021-06-07:
I have two problems below (both on local machine and docker):
TypeError: Caught TypeError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/rick/.local/lib/python3.6/site-packages/torch/utils/data/_utils/worker.py", line 202, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/rick/.local/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/rick/.local/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "train.py", line 525, in __getitem__
    dx = round(torch.normal(0, 2) * float(self.random_translation[0]))
TypeError: normal() received an invalid combination of arguments - got (int, int), but expected one of:
 * (Tensor mean, Tensor std, *, torch.Generator generator, Tensor out)
 * (Tensor mean, float std, *, torch.Generator generator, Tensor out)
 * (float mean, Tensor std, *, torch.Generator generator, Tensor out)
 * (float mean, float std, tuple of ints size, *, torch.Generator generator, Tensor out, torch.dtype dtype, torch.layout layout, torch.device device, bool pin_memory, bool requires_grad)

TontonTremblay, 2021-06-07:
Which pytorch version are you using? 
`round(torch.normal(0, 2)` -> round(torch.normal(0.0, 2.0))

Chi-Ruei, 2021-06-08:
I have tried
torch 1.6.0/1.5.1/1.5.0
torchvision 0.7.0 /0.6.0

Cuda 10.2

TontonTremblay, 2021-06-08:
Did you try the fixed I had above? Add `.0` to the digits inside the normal function.

Chi-Ruei, 2021-06-08:
No, it didn't.

I will try thank you

Chi-Ruei, 2021-06-08:
[image: Screenshot from 2021-06-08 13-27-00.png]

Chi-Ruei, 2021-06-08:
still have the same problem

TontonTremblay, 2021-06-09:
Maybe try torch 0.4.0

Chi-Ruei, 2021-06-09:
Thanks, I'll check it out.

Schramme793, 2021-06-15:
Hey, I have almost the same problems like @Chi-Ruei . I'm trying to execute the train.py to train a dataset which was generated with NDDS. I tried different versions of torch and up to this point I'm unable to install torch 0.4.0. The error I'm getting after executing the train.py script is the following:

python3 train.py --data /home/entwicklung/NDDS_Dataset/Inlet_orbital_randomized_2021.06.13-11.51.34 --object untitled --outf untitled --gpuids 0 --epochs 60 --batchsize 8
start: 10:30:21.838540
load data
training data: 13 batches
load models
Training network pretrained on imagenet.
Traceback (most recent call last):
  File "train.py", line 1398, in <module>
    _runnetwork(epoch,trainingdata)
  File "train.py", line 1336, in _runnetwork
    for batch_idx, targets in enumerate(loader):
  File "/home/entwicklung/.local/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 345, in __next__
    data = self._next_data()
  File "/home/entwicklung/.local/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 856, in _next_data
    return self._process_data(data)
  File "/home/entwicklung/.local/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 881, in _process_data
    data.reraise()
  File "/home/entwicklung/.local/lib/python3.8/site-packages/torch/_utils.py", line 395, in reraise
    raise self.exc_type(msg)
TypeError: Caught TypeError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/entwicklung/.local/lib/python3.8/site-packages/torch/utils/data/_utils/worker.py", line 178, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/entwicklung/.local/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/entwicklung/.local/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "train.py", line 525, in __getitem__
    dx = round(torch.normal(0, 2) * float(self.random_translation[0]))
TypeError: normal() received an invalid combination of arguments - got (int, int), but expected one of:
 * (Tensor mean, Tensor std, *, torch.Generator generator, Tensor out)
 * (Tensor mean, float std, *, torch.Generator generator, Tensor out)
 * (float mean, Tensor std, *, torch.Generator generator, Tensor out)
 * (float mean, float std, tuple of ints size, *, torch.Generator generator, Tensor out, torch.dtype dtype, torch.layout layout, torch.device device, bool pin_memory, bool requires_grad)

@TontonTremblay, do you know any other versions of torch above 0.4.0 which are working?

I'm looking forward to hear from you and hope you have any further advice.

TontonTremblay, 2021-06-15:
I think you need to update the call to torch.normal(0,2). 

```
        dx = round(np.random.normal(0, 2) * float(self.random_translation[0])) 
        dy = round(np.random.normal(0, 2) * float(self.random_translation[0]))
        angle = round(np.random.normal(0, 1) * float(self.random_rotation))
```
This used to be the code, you could change it back, should solve the problem. But the code might cause issues. I would suggest to simply debug the function and find the right types to input. `(float mean, Tensor std, *, torch.Generator generator, Tensor out)` right now we are sending int, I need to update the code.

TontonTremblay, 2021-06-16:
I just comited changes to the code so it would use floats. Can you tell me if this solves your issue?

Schramme793, 2021-06-16:
@TontonTremblay, thank you for your quick response. I just tested your changes a few minutes ago, but unfortunately I am still having problems within getitem. For further explanation, I am trying to run the training script with an test dataset of just 100 PNGs + Json's to see if it is working on my computer. In the next step I would like to test a similar dataset with 4000 PNGs and in the last step one with around 20000 PNGs. All datasets are generated by NDDS. 
The issues I am facing right after your changes are:

python3 train.py --data /home/entwicklung/NDDS_Dataset/Test_Dataset_Inlet --object untitled --outf untitled --gpuids 0 --epochs 10 --batchsize 1 --imagesize 400
start: 08:32:11.053665
load data
training data: 100 batches
load models
Training network pretrained on imagenet.
Traceback (most recent call last):
  File "train.py", line 1402, in <module>
    _runnetwork(epoch,trainingdata)
  File "train.py", line 1340, in _runnetwork
    for batch_idx, targets in enumerate(loader):
  File "/home/entwicklung/anaconda3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 336, in __next__
    return self._process_next_batch(batch)
  File "/home/entwicklung/anaconda3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 357, in _process_next_batch
    raise batch.exc_type(batch.exc_msg)
TypeError: Traceback (most recent call last):
  File "/home/entwicklung/anaconda3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 106, in _worker_loop
    samples = collate_fn([dataset[i] for i in batch_indices])
  File "/home/entwicklung/anaconda3/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 106, in <listcomp>
    samples = collate_fn([dataset[i] for i in batch_indices])
  File "train.py", line 529, in __getitem__
    dx = round(torch.normal(0.0, 2.0) * float(self.random_translation[0]))
TypeError: normal() received an invalid combination of arguments - got (float, float), but expected one of:
 * (Tensor mean, Tensor std, torch.Generator generator, Tensor out)
 * (Tensor mean, float std, torch.Generator generator, Tensor out)
 * (float mean, Tensor std, torch.Generator generator, Tensor out)
 
Thank you for your support.

TontonTremblay, 2021-06-16:
It should be fixed now.

TontonTremblay, 2021-06-18:
I just tested! it works.

closed 2021-06-18 by TontonTremblay