NVlabs/Deep_Object_Pose #174: Camera Intrinsic for HOPE 3D dataset

Jing-lun, 2021-06-09:
Hi there,

I am currently creating a HOPE dataset with a BOP format. However, it requires providing the camera intrinsic of the HOPE dataset. Does anyone know the specific parameters of that? 

Thanks so much!

TontonTremblay, 2021-06-09:
@swtyree do you remember where we stored BOP friendly HOPE models? The intrinsic should be in the `_camera_settings.json` file of the dataset.

TontonTremblay, 2021-06-09:
https://github.com/swtyree/hope-dataset/blob/master/preview.py#L91-L95 look here.

Jing-lun, 2021-06-09:
> https://github.com/swtyree/hope-dataset/blob/master/preview.py#L91-L95 look here.

Thank you for your prompt reply! I found the parameters you used!

```
  "cx": 326.28275,
  "cy": 245.73769,
  "depth_scale": 0.1,
  "fx": 607.1532,
  "fy": 608.33,
  "height": 480,
  "width": 640
```

TontonTremblay, 2021-06-09:
we used a realsense camera.

closed 2021-06-11 by TontonTremblay