NVlabs/Deep_Object_Pose #175: YCB-Video weights using NViSII

xEnVrE, 2021-06-17:
Dear @TontonTremblay,

I recently had the change to read the work "NViSII: A Scriptable Tool for Photorealistic Image Generation" you are one of the author of. I understood that you have re-trained DOPE using data containing YCB-Video objects generated using NViSII.

I wanted to ask whether you were planning to share the associated weights.

Thank you

xEnVrE, 2021-06-17:
> I only retrained the meat can. Would you like to try these weights? I can
upload them.

Yes, that would be very appreciated. Thank you!

TontonTremblay, 2021-06-17:
The weights are now in the google drive folder with all the other weights. `net_meat_shiny_14.pth`. I assume you know how to set it up with the code base, e.g., setting the cuboid size and pointing to the right path. 

This should be the direct link. https://drive.google.com/file/d/1ergPUxKDRoeIOvqWOpmWzoR-jmNk0ou1/view?usp=sharing

closed 2021-06-17 by TontonTremblay

xEnVrE, 2021-06-17:
Thank you @TontonTremblay