NVlabs/Deep_Object_Pose #181: Long time no see!

Ademord, 2021-07-22:
Hello, it's been a while since I worked with DOPE on the chairs, if ya'll might remember :)

I was wondering if you had any clue if I could use DOPE standalone of ROS and the 3D Datasets? 

I am working with Unity and reinforcement learning now, and I was thinking it would be cool to use DOPE in my current agent :) 

As a separate question, do you think DOPE could be exported as an ONNX model (does the model use [ONNX supported ops](https://github.com/onnx/onnx/blob/master/docs/Operators.md)?)) 

Let me know please!

TontonTremblay, 2021-07-23:
Hello, welcome back :P we missed you ;) 

I had exported dope network as onnx model a while back, like 3 years ago. You should not have any issues now. I had to hack some stuff if I remember correctly. 

Which 3D datasets are you referring to? We have used DOPE in a couple of RL papers (https://sites.google.com/view/guapo-rl, https://arxiv.org/pdf/1911.09233.pdf). Hope this can help you.  

John

Ademord, 2021-07-23:
haha thanks! feels a bit like home coming back :) (nostalgia) Hopefully when my masters is done I can dedicate more time to this repo as a hobby and less of "gotta do some project, let's try to propose something with this amazing tool made by amazing people :P" .
----
Ah I meant to separately export the DOPENetwork if there was anything I should be careful with (like removing an organ from a body, if there are connections I should be aware of).

Side note: I am in the middle of writing my theoretical background (chap 2 of thesis) so I might be a bit slow to actually put DOPE in the loop of my current implementation (~2 weeks of writing). 

So I am thinking of exporting [DOPENetwork](https://github.com/NVlabs/Deep_Object_Pose/blob/master/src/dope/inference/detector.py#L38) as onnx and feeding it some input and expecting some output.

Now that I think about it, an alternative is to export my Unity Environment to RLGym and try to pass the info DOPE needs as side channels, instead of trying to import DOPE into unity (ONNX, barracuda).

- how could I get informed what specific inputs and outputs are required? other than reverse engineering the network, maybe someone already asked how to just use it for inference? 🙏🏻  (as always, if anyone else has done a jupyter notebook for everyone to grasp it and plug it in, that would be the ideal universe).

Franco

TontonTremblay, 2021-07-23:
I have an internal version without any ros in it. send me an email jtremblay@nvidia.com I will send you a version of it.

closed 2025-08-15 by nv-jeff