NVlabs/Deep_Object_Pose #183: How to convert .pth files to ONNX

LTU-Eimantas, 2021-08-05:
So, I training my network, on my custom object for jetson NX to detect. So in order to get more performance, I use Nvidia Isaac, but it uses ONNX model. It has a script to convert from Torch model to ONNX one. Everything is working with your provided pre-trained wages, but when I trying to convert my wages I get an error. I notice that your wages are binary, mine is something like zip folder. How do I get that binary pth wages? Or do you have another way to convert to Onnx wages? I tried a few methods to save but still getting folder like wages, maybe I missing something very obvious, I am not sure... 

P.S. I am using your train.py file for training.

TontonTremblay, 2021-08-06:
https://discuss.pytorch.org/t/model-saving-as-a-zip-file/111130 

I think this is what is causing the issue. With new pytorch version, how the weights are saved and loaded is different. You might just have to do some research to figure out how to do onnx from there. Pytorch has been pretty good at being retro compatible, e.g. you can load the saved weights I shared with newer version. Hope this helps.

closed 2021-08-06 by TontonTremblay

LTU-Eimantas, 2021-08-07:
Hmm, ok I see. Thanks for the help. Maybe then I will try to convert it to ONNX from this new torch model, but in order to move one with a project (it's very time-sensitive), I will lower my PyTorch version.
One more thing. Not sure who is maintaining Nvidia ISAAC version, but I think you should suggest upgrading that converting script because ISACC uses PyTorch 1.6.0 version.  I am sure a lot of people would like to try to implement it in their AVG's with Jetson nano or NX.