NVlabs/Deep_Object_Pose #184: How to visualize synthetic data for DOPE

alishan2040, 2021-08-29:
Hi,

I'm working on generating synthetic data in order to train DOPE network. But when I plot the data, I'm not get correct bounding box around the objects, 3D bbox for example. Please see the example from the training set.
[image]

My code to display the 3D box is given below:
-------------------------------------------------------------
def draw_bbox_8_2D(path, color = (0, 255, 0), thickness = 2):
    """ Draws the 2D projection of a 3D model's cuboid on an image with a given color.

    # Arguments
        draw_img     : The image to draw on.
        bbox_8_2D    : A [8 or 9, 2] matrix containing the 8 corner points (x, y) and maybe also the centerpoint.
        color     : The color of the boxes.
        thickness : The thickness of the lines to draw boxes with.
    """
    f = open('{}.json'.format(path))
    data = json.load(f)
    rgb = "{}.rgb.png".format(path)
    draw_img = cv2.imread(rgb)
    bbox_8_2D = np.array(data['objects'][0]["projected_cuboid"])
    
    #convert bbox to int and tuple
    bbox = np.copy(bbox_8_2D).astype(np.int32)
    bbox = tuple(map(tuple, bbox))
    
    #lower level
    cv2.line(draw_img, bbox[0], bbox[1], color, thickness)
    cv2.line(draw_img, bbox[1], bbox[2], color, thickness)
    cv2.line(draw_img, bbox[2], bbox[3], color, thickness)
    cv2.line(draw_img, bbox[0], bbox[3], color, thickness)
    #upper level
    cv2.line(draw_img, bbox[4], bbox[5], color, thickness)
    cv2.line(draw_img, bbox[5], bbox[6], color, thickness)
    cv2.line(draw_img, bbox[6], bbox[7], color, thickness)
    cv2.line(draw_img, bbox[4], bbox[7], color, thickness)
    #sides
    cv2.line(draw_img, bbox[0], bbox[4], color, thickness)
    cv2.line(draw_img, bbox[1], bbox[5], color, thickness)
    cv2.line(draw_img, bbox[2], bbox[6], color, thickness)
    cv2.line(draw_img, bbox[3], bbox[7], color, thickness)
    
    #check if centerpoint is also available to draw
    if len(bbox) == 9:
        #draw centerpoint
        cv2.circle(draw_img, bbox[8], 3, color, -1)
    return draw_img
    
Can you spot the problem?
Thanks

TontonTremblay, 2021-08-30:
I am not sure what the problem is to start with. The picture you shared looks fine to me. Which tool are you using to generate the data?

18438627656, 2022-02-26:
> 我不确定问题是从什么开始的。 你分享的照片对我来说很好。 您使用哪个工具来生成数据？

Thank you very much for sharing, did you use NVDU visualization on FAT data sets?  I encountered an AttributeError: 'NoneType' object has no attribute 'set_text_color'  Can you help me?

closed 2025-08-20 by nv-jeff