NVlabs/Deep_Object_Pose #186: Need help with custom object training

LTU-Eimantas, 2021-09-21:
Hello, I really like the work you have done! But I need help with my model. This is my model.
[image]
[image]
[image]
[image]

So I trained my model with 90k images and 10k for testing which is made with NVISII and run it 60 epochs. It was painfully slow took about 7 days. I got these tensor images from the training algorithm. 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
and  I have run the save option and got these annotations. 
[image]
So after 60 epochs, I got about 0.01-0.009 loss. After running the interface, I got these tensor results 
[image]
[image]

But if I try to run on the real object I got nothing and this tensor map.
[image]

So do you have suggestions on how to improve it? It seems like some tensors are good but others don't make sense. One part says it should fix if I feed more data and train more, but it's real times consuming on my rtx 3060 12gb so I thought I might get a second opinion.

TontonTremblay, 2021-09-22:
These results are highly encouraging, the centroid is well detected. In your training data, do you think the camera field of view is similar to what you are using. Did you try to move away from the robot? It looks like your training data has the robot somewhat far. Also you should train with only a single instance. And since the robot is always going to be on the ground, you could make the poses not as random. I would say 1 or 2 more iteration(s) of data generation. For dope I probably had 10-15 iterations of data generation.  

I am sorry for the 3060 not delivering. I think epoch 20 or 30 should be enough to give you a good idea if it is working.

TontonTremblay, 2021-09-22:
can you share your 3d model, did you try with nvisii or used ndds?

TontonTremblay, 2021-09-22:
looking at it again, it seems like there is a symmetry on your model, e.g., the left and right side look similar, is that correct?

LTU-Eimantas, 2021-09-22:
> can you share your 3d model, did you try with nvisii or used ndds?

It's not full because my engineer team doesn't need all colored models so I added using nvisii and also some materials. 
https://drive.google.com/drive/folders/1p_PYmCjgyNaI-slY8rUb0s4JVfD4nJeH?usp=sharing

LTU-Eimantas, 2021-09-22:
> looking at it again, it seems like there is a symmetry on your model, e.g., the left and right sides look similar, is that correct?

yeah, they really similar just one button has a different color. But I manage to get some results from these trained wages, but from some perspectives it's isn't working maybe as you said I need to strict rotation axis and make more positions in depth for my training data.
[image]

TontonTremblay, 2021-09-22:
Yeah I would just limit the view of the object to one of it side, and do not let it go upside down. Also you can randomly color the robot. We did that in the robot pose estimation work and it helped. Make also the robot appear closer to camera. The testing image you share looks pretty good to me.

mintar, 2021-09-23:
Here's a script that will modify your existing dataset so that it only shows the object from one side:

https://github.com/NVlabs/Deep_Object_Pose/issues/176#issuecomment-877058216

TontonTremblay, 2021-11-15:
There is a thing on top of your robot (sorry I just saw this) You should model the thing on top of your robot. But the results are quite good. Good work :P

LTU-Eimantas, 2021-11-16:
> There is a thing on top of your robot (sorry I just saw this) You should model the thing on top of your robot. But the results are quite good. Good work :P

Yeah, I know it, but I don't need to detect it, only that base and its center because robots going to have different modules on top of it, and training that every time would be really times consuming and not practical.

TontonTremblay, 2021-11-16:
I am not saying add it to the pose estimation, but if you had it in the training data, the results would be more stable. But overall are you happy with the results?

LTU-Eimantas, 2021-11-16:
I think it's quite good ( of course could be better :D ), but as a network that is trained basically on synthetic data, I am quite impressed. Of course, further training will be done if the cost of implementation won't be too big, because even on my laptop with gtx 1050 it's struggling to run. Probably Nvidia jetson NX would be required.  But everything depends on the people above me :D .

sejmoonwei, 2021-12-01:
Hello @LTU-Eimantas .
[image]
Wow , your dataset looks pretty well . How do you deploy the jpg/png file as the background in vnisii? 
Could it be set by dome_light_texture manner ?I found it would cause a spherical deformation on the image on my end.

TontonTremblay, 2021-12-01:
I have been using dome light. https://github.com/owl-project/NVISII/blob/master/examples/17.materials_visii_interactive.py#L13-L14 

I have downloaded a pretty large set from https://polyhaven.com/hdris This is what I used in the https://arxiv.org/abs/2105.13962 I hope this helps. 

Otherwhise you can use the segmentation mask to do copy and paste onto normal images.

sejmoonwei, 2021-12-04:
> I have been using dome light. https://github.com/owl-project/NVISII/blob/master/examples/17.materials_visii_interactive.py#L13-L14
>
> I have downloaded a pretty large set from [...]

Thank you for helping me again . I'll check this method.

TontonTremblay, 2021-12-04:
Once I am back from vacation I will try to put a script together to share
here.

sejmoonwei, 2021-12-04:
Oh I would appreciate it if you can share a template script . Sorry to have disturbed the vacation , do enjoy it .🤣

TontonTremblay, 2021-12-13:
https://github.com/NVlabs/Deep_Object_Pose/tree/master/scripts/nvisii_data_gen Here is the script I promised. It is pretty bared. Feel free to send PRs.

sejmoonwei, 2021-12-17:
> https://github.com/NVlabs/Deep_Object_Pose/tree/master/scripts/nvisii_data_gen Here is the script I promised. It is pretty bared. Feel free to send PRs.
Thanks for sharing this work. Though I haven't had a chance to try this script  as I was looking into the loss term , I'll post there if any progress was made.

closed 2025-08-20 by nv-jeff