NVlabs/Deep_Object_Pose #188: DOPE in a Docker Container: Error when trying to run the container

markusiscoding, 2021-10-04:
Hello,
I am trying to run DOPE in a docker container and I followed the steps described here:
https://github.com/NVlabs/Deep_Object_Pose/blob/master/docker/readme.md

When trying step 4, I get the following result:
`$ ./run_dope_docker.sh
Container name     : nvidia-dope-v2
Host directory     : /home/jakobubuntu/dope
Container directory: /root/catkin_ws/src/dope
Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Get "http://%2Fvar%2Frun%2Fdocker.sock/v1.24/containers/json?all=1&filters=%7B%22name%22%3A%7B%22%5E%2Fnvidia-dope-v2%24%22%3Atrue%7D%7D": dial unix /var/run/docker.sock: connect: permission denied
Creating new DOPE docker container.
non-network local connections being added to access control list
docker: Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post "http://%2Fvar%2Frun%2Fdocker.sock/v1.24/containers/create?name=nvidia-dope-v2": dial unix /var/run/docker.sock: connect: permission denied.
See 'docker run --help'.`

So I tried adding "sudo" but then I got this result:
`$ sudo ./run_dope_docker.sh
Container name     : nvidia-dope-v2
Host directory     : /home/jakobubuntu/dope
Container directory: /root/catkin_ws/src/dope
Creating new DOPE docker container.
non-network local connections being added to access control list
docker: Error response from daemon: could not select device driver "" with capabilities: [[gpu]].
ERRO[0000] error waiting for container: context canceled`

How can I fix this problem?

TontonTremblay, 2021-10-05:
Can you tell us more about the machine you are using. It sounds like it
does not recognize your gpu. Are you using nvidia docker?

markusiscoding, 2021-10-05:
Thanks for your fast reply. 
This is the machine that I am using: 
[image]

I reinstalled the GPU driver(version: 470.74 )  and deleted and reinstalled nvidia-docker following this install guide:
https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/install-guide.html#docker

Now this is the response I get:

```
~/dope/docker$ sudo ./run_dope_docker.sh 
Container name     : nvidia-dope-v2
Host directory     : /home/jakobubuntu/dope
Container directory: /root/catkin_ws/src/dope
Creating new DOPE docker container.
non-network local connections being added to access control list
Unable to find image 'nvidia-dope:noetic-v1' locally
docker: Error response from daemon: pull access denied for nvidia-dope, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.

```

TontonTremblay, 2021-10-05:
@nv-jeff Do you think you could look into this please? Did you rebuild the docker image/container? 

Also I never used the mx 150, so I am not sure if they are limitation with the card.

nv-jeff, 2021-10-05:
First, try restarting your docker daemon; either reboot your machine or do:
`sudo service docker restart`
or
`systemctl restart docker`

markusiscoding, 2021-10-06:
I tried all three suggestions but I still get the same result as before. Is there anything else I can try?

TontonTremblay, 2021-10-06:
I am so sorry, do you have access to a different machine that is not a laptop. Gpus on laptops are great, but sometimes because of integrated rendering to save battery it is harder to run them correctly with deep learning system or dockers.

nv-jeff, 2021-10-06:
This is more of a docker issue than an issue with DOPE. Looking around, others have encountered this issue (https://github.com/NVIDIA/nvidia-docker/issues/1243) and have been able to solve it by installing a required package:
`sudo apt-get update
sudo apt-get install -y nvidia-container-toolkit
`

markusiscoding, 2021-10-07:
Apparently, the image wasn't built correctly because it only showed the nvidia/cuda repository. 
I tried building it again and it worked! 
Now I can proceed with trying to get DOPE to work for my project. Thanks for the quick support from all of you!
I also added my user to the docker group now - thanks for that tip as well.

closed 2021-10-07 by markusiscoding