NVlabs/Deep_Object_Pose #193: Loss gets stuck

javierrodenas, 2021-11-24:
Hello, 

I am experimenting the following issue:

[image]

It is getting stuck around 0.006-0.005. Does it happen to you? 

Thank you in advance

TontonTremblay, 2021-11-24:
I am not sure, what sort of GPU do you have? Which data are you using?

javierrodenas, 2021-11-25:
I am using the 2080 RTX Ti (12 GB).

On the other hand, I am using this data currently  (some examples):
[image]
[image]

My experience with this environment is that it does not matter which data I use, the loss is always around 0.005

javierrodenas, 2021-11-26:
I was testing different parameters with same type of data:

Green: learning_rate 0.0001 and batch_size 16
Blue: learning_rate 0.001 and batch_size 8

[image]

The result of train loss is more of less the same and it gets stuck around 0.005.

Any recommendation? Do you experimented the same?

Thank you in advance!

TontonTremblay, 2021-11-26:
It might be caused by the slightly symmetrical object.

javierrodenas, 2021-11-29:
I can try with non symetrical object.

Another possible issue here could be the color. I comment that because in another github issue I saw a guy who was playing with different color cubes #128. Could my color be an issue? It is white/light grey.

I removed the noise and the loss got to 0.003 (It decreased from 0.005 to 0.003) but it is not able to detect the piece during the predictions

TontonTremblay, 2021-12-01:
can you share how the belief maps look like?