NVlabs/Deep_Object_Pose #199: If the input is a picture, how to estimate the pose of the target

single-pup, 2021-12-21:
Hello, I don't know much about ros. I found that the inference file contains functions for pose estimation. When the input is a picture, how can I use these existing functions to estimate the pose of the target in the picture? Thank you for your help!

chebisch, 2021-12-21:
DOPE does not directly estimate the pose of an object, but the location of the vertices of the object's bounding box in the 2D image. With known camera intrinsics and object size, you can then estimate the object's 3D pose using the perspective-n-point (PnP) algorithm (see [here](https://en.wikipedia.org/wiki/Perspective-n-Point)).

single-pup, 2021-12-27:
Hello, I just started learning deep learning. My teacher and I have a strong interest in DOPE and believe that it has great learning value for our subject. Recently I studied your open source code and modified it slightly to meet our subject. Can you share some data sets that you use for training (for example, pictures and label files containing cookies, corn, etc.) so that I can test the code and make my own data set. If you offend, please forgive me and thank you for your help!

TontonTremblay, 2021-12-28:
You can look at the issues different people shared dataset. And you can
look at the FAT dataset as well. Recently I have uploaded a new version of
the training script and inference. As well as that I have uploaded scripts
to generate synthetic data for training dope train2. I hope this helps.
Check the news on the main Readme.