NVlabs/Deep_Object_Pose #200: DOPE implementation on AUBO robotic arm

Rinvay67, 2021-12-21:
Now I have successfully deployed DOPE on the machine, and I can get the "/dope/pose_cracker" topic under the "camera_color_optical_frame". Also, I have successfully got the transform from the "base_link" to "camera_color_optical_frame", How can I get the transform of the object under the "base_link" ?

TontonTremblay, 2022-02-10:
you have to run a static transform from your camera to your robot. If you have it then you can querry the poses. Does that make sense?

edvart-ros, 2022-04-07:
This is how i publish the transform for my setup. Im not a ros-expert but this seems to do what u want. DOPE calculates the pose of the object from the optical frame of my camera, so all i have to do is broadcast that transform along.

`#!/usr/bin/env python3

import roslib
import rospy
from geometry_msgs.msg import PoseStamped
import tf

def pose_callback(msg):
    position = msg.pose.position
    orientation = msg.pose.orientation

    br = tf.TransformBroadcaster()
    br.sendTransform((position.x, position.y, position.z), (orientation.x, orientation.y, orientation.z, orientation.w), rospy.Time.now(),"cheezit","camera_color_optical_frame")
    rospy.loginfo('tf published')
if __name__ == '__main__':
    rospy.init_node('cheezit_tf_broadcaster')
    rospy.Subscriber('dope/pose_cracker', PoseStamped, pose_callback, queue_size = 10)
    rospy.spin()
`

mintar, 2022-04-07:
It's better to use the `detected_objects` topic instead of the `pose_*` topics (like `pose_cracker`), especially if you have multiple objects of the same type. The poses inside the `detected_objects` message are in the optical frame, and you can create a `tf2.TransformListener` and call its `transformPose` method to transform them into an arbitrary tf frame, like `base_link`.

edvart-ros, 2022-04-07:
> It's better to use the `detected_objects` topic instead of the `pose_*` topics (like `pose_cracker`), especially if you have multiple objects of the same type. The poses inside the [...]

Awesome! I've installed this package just today, so this is great to know.