NVlabs/Deep_Object_Pose #203: DOPE not training

Myrmecoman, 2022-01-03:
Hello,

I am opening an issue here since I cannot figure out why DOPE is not learning on my side. I'll provide here as much information as possible about my system and the steps I did to launch the learning process.

# System specifications

I have 1 single RTX3070. I am using Ubuntu 20.04. My nvidia driver version is 495.44, with CUDA 11.5. I have torch 1.10.1+cu113.

# How I generated my training data

I used the script generate_dataset.py and generated 45000 images. -spp was set at 3800, with 0 distractor and 1 object per image since it is written here to train with one single instance (https://github.com/NVlabs/Deep_Object_Pose/issues/186#issuecomment-924521649). Here are a few examples of what I have :

[image]

[image]

I also have some images where the object is partially visible, or not visible at all. Does it matter though ?

[image]

[image]

# The training

I used train.py provided in scripts/train2. Here is the command I used :
python3 -m torch.distributed.launch --nproc_per_node=1 train.py --network dope --epochs 60 --batchsize 12 --outf tmp/ --data ../nvisii_data_gen/output/dataset/ --verbose --lr 0.0000001 --pretrained False

First thing to note here is the very low learning rate. I chose this one since a bigger number achieves a loss smaller than 0.001 in only a few minutes, well before the end of the first epoch. I hoped with adam it would correct itself anyway, which I think it did.

I often (but not always) have a message printing in the console before each epoch which reads "WARNING:root:NaN or Inf found in input tensor.". This message comes from the line 519 in train.py "writer.add_image(f"{post}_belief_ground_truth_{i_output}", grid, epoch, dataformats="CWH")". Could this tell anything about the IA not learning ?

Now about the loss. I tried to train on a small sample (7100 images) for 10 epochs, it does converge but towards nothing apparently. The loss should not get this low in the end of the training, and also for some reason on the first iteration the loss is already very very low despite the parameter  "--pretrained False".

```

  warnings.warn(
start: 08:29:26.797602
load data: ['../nvisii_data_gen/output/dataset/']
load data: 
training data: 1184 batches
load models
ready to train!

Train Epoch: 1 [0/14200 (0%)]	Loss: 0.003868995700032
Train Epoch: 1 [1200/14200 (8%)]	Loss: 0.003753516823053
Train Epoch: 1 [2400/14200 (17%)]	Loss: 0.003745390567929
Train Epoch: 1 [3600/14200 (25%)]	Loss: 0.003749598748982
Train Epoch: 1 [4800/14200 (34%)]	Loss: 0.003582739736885
Train Epoch: 1 [6000/14200 (42%)]	Loss: 0.003265241160989
Train Epoch: 1 [7200/14200 (51%)]	Loss: 0.003937990870327
Train Epoch: 1 [8400/14200 (59%)]	Loss: 0.003569062333554
Train Epoch: 1 [9600/14200 (68%)]	Loss: 0.003321033669636
Train Epoch: 1 [10800/14200 (76%)]	Loss: 0.003342077136040
Train Epoch: 1 [12000/14200 (84%)]	Loss: 0.003121480811387
Train Epoch: 1 [13200/14200 (93%)]	Loss: 0.002714241854846

Train Epoch: 2 [0/14200 (0%)]	Loss: 0.002526555908844
Train Epoch: 2 [1200/14200 (8%)]	Loss: 0.001830663997680
Train Epoch: 2 [2400/14200 (17%)]	Loss: 0.001757183810696
Train Epoch: 2 [3600/14200 (25%)]	Loss: 0.001754474593326
Train Epoch: 2 [4800/14200 (34%)]	Loss: 0.001572249457240
Train Epoch: 2 [6000/14200 (42%)]	Loss: 0.001561113982461
Train Epoch: 2 [7200/14200 (51%)]	Loss: 0.001620477065444
Train Epoch: 2 [8400/14200 (59%)]	Loss: 0.001243051257916
Train Epoch: 2 [9600/14200 (68%)]	Loss: 0.001000139745884
Train Epoch: 2 [10800/14200 (76%)]	Loss: 0.001292466418818
Train Epoch: 2 [12000/14200 (84%)]	Loss: 0.001241447054781
Train Epoch: 2 [13200/14200 (93%)]	Loss: 0.001317200600170

Train Epoch: 3 [0/14200 (0%)]	Loss: 0.001491746865213
[... 95 lines cut ...]
Train Epoch: 10 [0/14200 (0%)]	Loss: 0.001038409769535
Train Epoch: 10 [1200/14200 (8%)]	Loss: 0.001347974292003
Train Epoch: 10 [2400/14200 (17%)]	Loss: 0.001235856907442
Train Epoch: 10 [3600/14200 (25%)]	Loss: 0.000962803722359
Train Epoch: 10 [4800/14200 (34%)]	Loss: 0.001052047358826
Train Epoch: 10 [6000/14200 (42%)]	Loss: 0.000654937815852
Train Epoch: 10 [7200/14200 (51%)]	Loss: 0.001176543184556
Train Epoch: 10 [8400/14200 (59%)]	Loss: 0.001152561511844
Train Epoch: 10 [9600/14200 (68%)]	Loss: 0.001249288558029
Train Epoch: 10 [10800/14200 (76%)]	Loss: 0.001024243887514
Train Epoch: 10 [12000/14200 (84%)]	Loss: 0.001011104555801
Train Epoch: 10 [13200/14200 (93%)]	Loss: 0.000892798765562
end: 10:37:35.776622

```

# The results

No surprise here, the belief map shows no learning at all. Here is the belief map for this input image after 60 epochs on a dataset of 45000 images :

[image]

[image]

So what did I do wrong ? Image generation took 3 days, and learning took another 3 days, I hope at least image generation is ok at this point.

TontonTremblay, 2022-01-03:
I am sorry you are having a hard time with the training process, I hope I can help you out. How does the tensroboard look like. I would say your drill appear to be somewhat close to the camera. In the tensorboard generated you can look at some testing images and their output. Can you share with me some screenshot or the file?

The loss looks like it is doing the right thing. The data you generated looks pretty good to me.

Myrmecoman, 2022-01-04:
I tried again in rviz with different positions, rotations and distances of the drill without much success.

I tried exporting my tensorboard data but did not manage to, so I will post here what it contains.
Here are the plots I get when loading tensorboard in the tmp/ folder generated by the training script :

[image]

Here are several testing images and their output in tensorboard, with input image, ground truth and belief guess :

Step 15 :

[image]
[image]
[image]

Step 20 (this one is strange, there are 2 drills apparently) :

[image]
[image]
[image]

Step 34 :

[image]
[image]
[image]

Step 43 :

[image]
[image]
[image]

Step 60 (the drill is not visible on this one) :

[image]
[image]
[image]

It looks like the AI learned a little on step 34, but then lost everything since the belief maps are really not good after that.
The graphs do not correspond to the AI performance apparently, the belief is not converging at all.

TontonTremblay, 2022-01-04:
I would say your data does not look too good. The process of generating data is very time consuming, I am sorry about that. I would be surprised that you would get it right on the first try. If it takes too long generate data with lower sample per pixel. I would add a few more instances to start with, there are a lot of empty images in there. I would also change the scale of the object. https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/nvisii_data_gen/single_video_pybullet.py#L232 change the scale to something like 0.75. Add a little bit more instances. Increases the likelyhood to have the object in the field of view. And I would add distractors as well. Use the google scanned download script (you do not have to download the whole thing).

Myrmecoman, 2022-01-05:
Ok I will do that thanks. I will keep this thread updated about how it goes.

Myrmecoman, 2022-01-06:
Ok so I generated 11600 images, reducing the drill scale, reducing the --box_size to 0.25 instead of the default 0.5, using 4-5 objects and 6-8 distractors. I trained this tonight (for about 15h), it reached epoch 45. I used the --objects arguments to tell that I only want to detect the screwdriver.

I'm providing here my results. Does it look promising before i generate more images and train it for longer ? I feel like it already converged and I won't get anything better.

[image]

Epoch 2 :

[image]
[image]
[image]

Epoch 15 :

[image]
[image]
[image]

Epoch 20 :

[image]
[image]
[image]

Epoch 26 :

[image]
[image]
[image]

Epoch 39 :

[image]
[image]
[image]

Epoch 43 :

[image]
[image]
[image]

Epoch 45 :

[image]
[image]
[image]

Thanks !

TontonTremblay, 2022-01-06:
This looks promising. I would continue training as is and you would have a
nice baseline to work with.

Myrmecoman, 2022-01-06:
Ok great, I'll generate 90000 images and train on 20 epochs this week-end. I'm crossing my fingers.

TontonTremblay, 2022-01-06:
Just train on what you have. I don't think you need that many images. 20k
to start will be more than enough.

Myrmecoman, 2022-01-10:
I trained the AI this weekend on a dataset of 22600 images, for 95 epochs. The results are better now, it looks to be extracting every black object though currently. Here is the raw image along with its belief maps :

[image]

[image]

Some of my images have a lot of occlusion, and I have one specific background (see image) which is pretty dark. Maybe I should remove it, take out some distractors and objects to have less occluded objects, and regenerate some images. I'll try it also with better spp. I might finally come to something I think.

[image]

I'll be closing this post and will keep this thread updated on my results if they get better, and if I keep working on it since I have a lot of other things to work on.
Thanks for your help, and I hope this thread will help others having the same difficulties as me. :+1:

mintar, 2022-01-10:
Just chiming in to say that this is not normal. DOPE should do way better than this; I've never seen belief maps that bad, and your training data looks very good. I don't know what the problem is, but there must be something wrong in the dataset annotation files or the DOPE config. Don't try to fine-tune yet (removing the dark background, adding more images, ...) before you found the problem.

Please double- and triple-check your DOPE config and the training data. You can also upload a small portion of the dataset (100 imgs or so, along with the `.json` files) and your DOPE config. I'll try to have a look if I can find the time.

Myrmecoman, 2022-01-10:
I verified by hand several annotations concerning bounding_box_minx_maxx_miny_maxy for distractors and objects, and they are correct. I saw some people drawing the bounding boxes on the images on other posts, but do not know how to achieve this. My dope config file is correct I think, it's loading the correct weights when I launch the ROS module.

Here is provided a link containing 100 images along with their json and exr files, the dope config I use, the weights I used in the previous post and my 3D model. The whole thing weights around 300MB (mostly the weights).

https://drive.google.com/drive/folders/1JKp2KdeMXUH9l5Os_AMebPRCqJ__J8V_?usp=sharing

Thanks for your interest on my issue.

mintar, 2022-01-10:
Could you add the `_object_settings.json` and `_camera_settings.json` files?

Myrmecoman, 2022-01-10:
There is a camera_info.yaml file in the config folder (which I did not modify for now), but I cannot find any _camera_settings.json or _object_settings.json. Wherever they are, I never modified those.

Aren't the bounding boxes generated using the centroïd of the collision meshes by NVISII ?

mintar, 2022-01-10:
Ok, never mind. These two files are only needed by the "old" training script (`src/train.py`) and by [nvdu_viz](https://github.com/NVIDIA/Dataset_Utilities/). I'll try to cobble these files together myself from the available data and visualize the stuff using nvdu_viz, and check if the bbox corners match the projected bbox corners in the json file (I think @TontonTremblay mentioned that he might have switched something up there). If that looks ok I'll try if training using the old `train.py` works. But won't get to that today I'm afraid.

Myrmecoman, 2022-01-10:
Don't worry I'm not in a hurry. Thanks for your help !

TontonTremblay, 2022-01-10:
@Myrmecoman could should also share what the training heatmaps look like. I would say that you probably need more training, but 95 epoch though should do it though. So yeah there might be a problem somewhere. 
- Did you try to train on a single image? 
- There is an other post where we discussed the keypoint orientation and how they are in a different order than the new one. If you look at the inference code without ros you should see it. 
- Do you resize your image before running the inference on it?
- The training 2.0 is not quite compatible with the ros version, I should work on making them compatible... sorry about that.

Myrmecoman, 2022-01-10:
What do you mean by training heatmap ? There is no such thing generated by the train2/train.py. I'm sorry i'm pretty much new to deep learning.
* I did not try to train on a single image, but it's definitely worth a try, i'll do it tomorrow when I have access to the computer again.
* I did not know about this issue. I'll definitely try to run the inference code without ROS. I have my doubts about it to be the source of the issue since the tensorboard already shows that the network is not trained correctly, but who knows.
* I do not resize my image, at least I did not change the code to do so, however now that you mention it I think the training data is 500x500p when the network requires 400x400p. I will maybe downscale the image to 400x400p just to be sure.

> @Myrmecoman could should also share what the training heatmaps look like. I would say that you probably need more training, but 95 epoch though should do it though. So yeah there might be a problem [...]

TontonTremblay, 2022-01-11:
training heatmaps, meaning training for longer. Sorry my words were a little confusing. Can you show me heatmaps from the tensorboard?

Myrmecoman, 2022-01-11:
Sure thing, here are the maps for this run :

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Myrmecoman, 2022-01-11:
Ok there is something wrong with the ground truth, when training on the same image it's changing overtime :

[image]

[image]

[image]

mintar, 2022-01-11:
Yep, there's definitely something wrong there. The ground truth shouldn't change. Also, the object is in the top left corner, so all white dots in the ground truth should also be clustered around the top left corner.

Myrmecoman, 2022-01-11:
Let me provide the command to reproduce and the data used :
```
python3 -m torch.distributed.launch --nproc_per_node=1 train.py --network dope --epochs 10000 --batchsize 12 --outf tmp_singleimg/ --data ../nvisii_data_gen/output/singledata/ --verbose --lr 0.0000001 --pretrained False --imagesize 500 --dontsave
```

--dontsave is necessary otherwise you may not have enough storage (200MB per epoch for the weights).

Link to the image and it's data files :
https://drive.google.com/drive/folders/1nINcv5EEGDhFdo_jLzZS2N1MA3HqkX9C?usp=sharing

mintar, 2022-01-11:
Hmm, the random crop would explain why the points move from frame to frame. They seem to move together, which is what would be expected by random crop. And a rotation of 180° could move them from the top left to the bottom right corner, but there seems to be some mirroring going on as well.

@Myrmecoman : Could you post the belief maps without the randomization?

Myrmecoman, 2022-01-11:
I'll do it tomorrow as I don't have access to the computer anymore. I will remove all these functions, however I'll still have to resize the image to 400x400p, or generate my images directly using this resolution.

TontonTremblay, 2022-01-11:
### Only random crop
I used the image and data I generated for the github with the script. I removed all the augmentation and kept the crop. I started to see belief maps around 200 epochs. Here is the command I used: 
```sh
python -m torch.distributed.launch --nproc_per_node=1 train.py --network dope --epochs 20000 --batchsize 32 --outf tmp2/ --data ../nvisii_data_gen/output/output_example/ --dontsave
```

Here are some results
[image]
[image]
[image]

### All augmentations
[image]
It is working as well, but in the process of training still. 

I would need to look at your data more closely. It probably just need more training. Make sure you are comparing image from the epochs, I noticed that tensorboard sometimes have different epochs for input, output, gt. So it sort of show that it is doing something crazy. Something I noticed is that you need to filter which object you want to train on. https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/train2/train.py#L95-L96 So make sure you only include your drill. I think it gets added using the substring. https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/train2/utils_dope.py#L222-L225

Myrmecoman, 2022-01-11:
I will try to reproduce your results tomorrow as well, using the same data and same command. I did filter the objects to train on, I should indeed verify if it's really the case in the code.
Well if I can reproduce your (almost) exact results, I will indeed just need to train more, maybe the object containing lots of curvatures makes it a hard time to train on ?

Myrmecoman, 2022-01-12:
Well you were right, I have the same results as you, I suppose I just need to train longer after all.

[image]

So I guess my drill simply is pretty hard to learn. I have generated in the meantime 50 000 images of 2 drills with no distractor. I'll run that on 100 epochs and see how it goes. I'll post news when results come. Thanks both for your help !

Myrmecoman, 2022-01-17:
Hello,

So after 3 days for 50 000 images and 5 days of learning, I have promising results. The 3D box does not always look right especially when the visible surface of the drill is small, but well its working finally. I think it might be smarter to only learn the top of the object for example, since it is less likely to be occluded by a hand, it is the part which contains the most texture contrasts and its shape is not as complex as the whole drill.

[image]

[image]

[image]

So yeah still having some issues but I guess it can all be made more robust with more images and learning.

mintar, 2022-01-17:
The belief maps look good now, congratulations! You can stop focusing on the training now. Instead, on to the next steps:

* It looks like you're using unrectified images. You should rectify the images before feeding them to DOPE, and make sure that your camera is well-calibrated and that the camera matrix that you give DOPE is correct.
* In the DOPE config that you posted a while ago, you specified the cuboid dimensions as "10.0 10.0 10.0". It seems you are still using that, because the box published by DOPE looks like it's a cube. The correct bbox dimensions seem to be "46.615639 42.882923 16.224327" according to your mesh (obtained using meshlab). Is the scaling correct? What units are those? 46 cm seem a bit much for your object. So perhaps you have to scale your mesh appropriately. Also, the reference coordinate system of your mesh is not in the bounding box center; you can probably save yourself a lot of trouble by moving it to the center (otherwise there is probably going to be a fixed offset somewhere).

Note that the camera parameters and bounding box size are not just for visualization - these are integral to the algorithm, because DOPE uses OpenCV's solvePNP function to move from the belief maps (which look good) to the pose estimation (which looks horrible in your data).

Myrmecoman, 2022-01-17:
- I do use unrectified images indeed for now. My principal concern was getting good belief maps, which is now the case.
- Yes the cuboid dimensions were absolutely arbitrary, I though it would not impact pose estimation. Thanks for pointing it does.
- The dimensions do not correspond to reality indeed. This is a 3D scanned model and the dimensions were already off when I got the model. However I use Blender, which is more of an artist thing than a true engineering software, so I did not know the true dimensions of my object. I will redimension according to reality.

Thanks for your comment. :+1:

mintar, 2022-01-17:
The model is really good (apart from the scaling)! Out of curiosity, what did you use to 3D scan it?

Myrmecoman, 2022-01-17:
I did not do it personally. I went to a company specialized in 3D scanning and 3D printing. They used a 3D hand scanner, which costs about 8000 euros, and which works using the deformation of projected light stripes on the object.The accuracy is pretty stunning, they said about 50 micron, even though it also depends on the number of polygons you want. So they did both sides and merged the 2 halves using their software. In the end there is still a few missing polygones like in the screw holes. A simple algorithm can easily fix them. In the end the model weights about 1GB, but I had memory errors with NVSII while generating the collision mesh so I resized the model at about 8MB and it worked.

TontonTremblay, 2022-01-18:
RE 3D models, we are working a on simpler solution to get a ruff 3d model. Hopefully I will be able to share the code soon. Congrats on getting everything to run. It takes a while to train and generate everything. I am also looking into making this a little bit more accessible.

Manu752, 2025-06-11:
How many epochs did you have to train for before you started seeing results? I am 290th epoch and this is my belief map -

[image]

[image]