NVlabs/Deep_Object_Pose #209: Running DOPE on ROS Melodic. cv_bridge python version error

bmunkhtulga, 2022-01-18:
Hello. I am trying to run DOPE on ROS Melodic with Ubuntu 18.04 (Because our robot does not have an official repo in ROS Noetic). I followed the instruction and everything went well. But when I launch dope using this line `roslaunch dope dope.launch`, I encountered a cv_bridge error. 

> [ERROR] [1642504011.301276]: bad callback: <bound method Subscriber.callback of <message_filters.Subscriber object at 0x7fac42f6e4e0>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/message_filters/__init__.py", line 76, in callback
    self.signalMessage(msg)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/message_filters/__init__.py", line 58, in signalMessage
    cb(*(msg + args))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/message_filters/__init__.py", line 225, in add
    self.signalMessage(*msgs)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/message_filters/__init__.py", line 58, in signalMessage
    cb(*(msg + args))
  File "/home/default/catkin_ws/src/dope/nodes/dope", line 228, in image_callback
    img = self.cv_bridge.imgmsg_to_cv2(image_msg, "rgb8")
  File "/opt/ros/melodic/lib/python2.7/dist-packages/cv_bridge/core.py", line 163, in imgmsg_to_cv2
    dtype, n_channels = self.encoding_to_dtype_with_channels(img_msg.encoding)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/cv_bridge/core.py", line 99, in encoding_to_dtype_with_channels
    return self.cvtype2_to_dtype_with_channels(self.encoding_to_cvtype2(encoding))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/cv_bridge/core.py", line 91, in encoding_to_cvtype2
    from cv_bridge.boost.cv_bridge_boost import getCvType
ImportError: dynamic module does not define module export function (PyInit_cv_bridge_boost)

After searching the source of error, I understood that cv_bridge is built with python2. [https://stackoverflow.com/questions/49221565/unable-to-use-cv-bridge-with-ros-kinetic-and-python3](url)
[https://linuxtut.com/en/01945ecea97c26ff8324/](url)

Similar to these URLs, I have tried building cv_bridge with python3 and failed. Did anybody make DOPE run in ROS Melodic? Should I just go back to ROS Kinetic?

closed 2022-01-18 by bmunkhtulga

TontonTremblay, 2022-01-18:
python 2/3 and ros is just a nightmare to deal with. Lots of hours lots of that. I feel like I cannot update my robot machine ever again to make sure I can still work with the robot :(. 

We ported the code to ros2 https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_pose_estimation which might be easier, but I do not have experience with it.

bmunkhtulga, 2022-01-19:
Yeah, it is really a nightmare. Lots of hours went into errors and bugs of version conflict. Good thing that ROS2 came out, but with my robot, I think I am not going to use it anytime soon. 

Regardless, thank you for providing DOPE and helping our issues.

TontonTremblay, 2022-01-19:
You beat me to the punch, but you are welcome ;)