NVlabs/Deep_Object_Pose #221: ERROR:Unable to find model weights:'weights_dope/resnet_simple/alphabe_soup.pth'

eugeneYz, 2022-02-27:
Hello, I followed the introduction ,downloaded the weights , saved them to the weights folder and unzip the weights.When I run the inference.py, it didn't seem to work. Can you help me to fix the bug?

eugeneYz, 2022-02-28:
Hi,I changed the weights path in the config_pose.yaml and run the inference.py.Now it happens to have new error:`
Traceback (most recent call last):
  File "inference.py", line 338, in <module>
    dope_node = DopeNode(config)
  File "inference.py", line 128, in __init__
    self.models[model].load_net_model()
  File "inference/detector.py", line 238, in load_net_model
    self.net = self.load_net_model_path(self.net_path)
  File "inference/detector.py", line 251, in load_net_model_path
    net = ResnetSimple()
NameError: name 'ResnetSimple' is not defined
`
I can't find some defintion or useful information about  'ResnetSimple'. 
Thank you.

TontonTremblay, 2022-02-28:
Did you train your model with the new training script, it is not compatible with the ros node yet. I think there is an other issue that had similar problems. I am sorry about this. New is not always better :(, I shared the script to simplify data generation. If I have time this week I can try to push a hot fix to have both versions on the ros node, but I would not count on it for now. I am happy to accept a PR for this though ;)

eugeneYz, 2022-03-01:
Sorry i'm a little confused. Do you mean the inference script (without ROS) should be run after training the model?