NVlabs/Deep_Object_Pose #232: need help --custom object training

HiHiAllen, 2022-03-18:
HI, i have  some doubt about my custom object training .Using NDDS, i generated  20k  images to train the net and run it  38 epochs (from epoch 20-38 , the loss fluctuates around  0.0018 ， so i shut it down )
Then i input several images from those  20k  images , and i got their beliefs maps ,it seems quite good ,i strongly believe that the training works.
Here are some belief maps:
[image]
[image]
[image]
[image]
[image]

BUT ,when i input some images photographed by my camera (astra pro  640*480), target cant be identified .
Here are some belief maps:
 
[image]
[image]
[image]

so,whats your opinion  , do i need more images to train the net ?  is 50k enough?  or some else problems?  Thanks for any help.

TontonTremblay, 2022-03-21:
It looks like there might be a symmetry on the object. I cannot see very well from my phone. But looking at the output from the belief maps I would think there is a sym.

mintar, 2022-03-22:
On the PC, it doesn't look like there is a rotational symmetry, only a mirror symmetry which should be fine.

HiHiAllen, 2022-03-25:
> It looks like there might be a symmetry on the object. I cannot see very well from my phone. But looking at the output from the belief maps I would think there is

HiHiAllen, 2022-03-25:
thanks , i wil try to train a new one to make a comparison.

closed 2022-03-25 by HiHiAllen