NVlabs/Deep_Object_Pose #233: how to generate the  Accuracy-threshold curves

DOPE-liyulong, 2022-03-22:
Hello, first of all, thank you for your work. Could you please explain in detail how to generate the  Accuracy-threshold curves in the paper?Why is there only one loss value after training?

TontonTremblay, 2022-03-22:
https://plotly.com/python/roc-and-pr-curves/ here is an example on how you can make your own curves.

DOPE-liyulong, 2022-03-23:
> 我不确定你需要什么。 您需要为 ADD 编写代码吗？ 我可以分享我在不同论文中使用的版本。

Thank you very much for your timely reply. Maybe I didn't make it clear. I know you used YCB-Vedio-Toolbox when making the ADD curve in dope paper. Do you use any other code besides that? If so, can you share this part of the code? Thank you again for your contribution. It has helped me a lot.

TontonTremblay, 2022-03-23:
https://drive.google.com/drive/folders/1vfajGlKj7faZw26UDFx8y4VT5mPYRtvR?usp=sharing 

Here are the new scripts I wrote afterwards for the nvisii papers. Check the notes.txt. I think I have everything in there for you to run it.

TontonTremblay, 2022-03-23:
You will have to download the hope dataset and fix the paths in the parsers.

DOPE-liyulong, 2022-03-23:
> https://drive.google.com/drive/folders/1vfajGlKj7faZw26UDFx8y4VT5mPYRtvR?usp=sharing
> 
> Here are the new scripts I wrote afterwards for the nvisii papers. Check the notes.txt. I think I have everything in there for you to run it.

Thank you for your prompt reply. I'll take a good look at what you shared. Thanks for sharing, it helped me again.

kooshyarkosari, 2022-06-15:
Hi There 

I want to generate Accuracy-threshold curves based on YCB- Video Dataset as you did in the paper, so is it possible to use the code that you mentioned in the previous comments for the Ycb - dataset instead of the Hope dataset?

Thanks in advanced

TontonTremblay, 2022-06-15:
Technically yes. But the script is not the easiest to work with. I can make an example to work with the hope-dataset structure. What do you think?

kooshyarkosari, 2022-06-15:
Yes, please, I really appreciate your help