NVlabs/Deep_Object_Pose #237: How can I get files like .obj and .png？

single-pup, 2022-03-26:
I tried to create a simple cuboid in 3DMAX and textured its surface. I can export the .max file and its corresponding .png file, and then the .obj file separately. But after the .obj files and .png files obtained by this method are run in the code of nvisii, my model is not included in all the output pictures.

TontonTremblay, 2022-03-26:
Do you mean you do not have a textured object? Or you do not see it on the
rendered images at all? Can you share some images for context?

single-pup, 2022-03-27:
[model.zip](https://github.com/NVlabs/Deep_Object_Pose/files/8356799/model.zip)This is the model I created in 3DMAX. After running nvisii's code, I didn't find my model in the output image.

TontonTremblay, 2022-03-27:
Did you try playing with the scale? Or running it interactive? I should add
debug steps to the read me? I will look into the model tomorrow.

single-pup, 2022-03-27:
Thanks for your help, it would be helpful if you could add debug to show the cause of the error. I recreated the obj file in SOLIDWORKS and it appears in the output image, but the texture of model are not showing. I'm not sure how to generate the png file, can you tell me how you generate obj and png？

TontonTremblay, 2022-03-27:
There is a debug version. —interactive

TontonTremblay, 2022-03-27:
In your `.mtl` change this line: 
```
map_Kd F:\work\model\model\model.png
```
to 
```
map_Kd model.png
```
Make sure you have that in your .obj 
```
mtllib model.mtl
```

TontonTremblay, 2022-03-27:
Your export does not look too good, did not have time to look into loading it into nvisii yet.

mintar, 2022-03-28:
Apart from the problems that @TontonTremblay already mentioned, there's one more: your model doesn't have UVs (texture coordinates). If you open up your model, you see there are no lines starting with `vt`. Also see: https://en.wikipedia.org/wiki/Wavefront_.obj_file

single-pup, 2022-04-07:
Thank you for the help! I recreated my model and successfully read my model using nvisii. I entered
 `python3 single_video_pybullet.py --width 1920 --height 1080 --objs_folder (path to myobj) --nb_objects 1 --nb_distractors 5`
 to generate a dataset of 2000 images. 
I am not using camera configuration.I modified [https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/train2/train.py](url)来训练。
Since the size of my images is 1920*1080, I have modified the code from
`transform = transforms.Compose([
        transforms.Resize(opt.imagesize,
        transforms.ToTensor()])`
to
`transform = transforms.Compose([
        transforms.Resize((1080,1920)),
        transforms.ToTensor()])`
and entered
` python train.py --data  (path to mydataset) --objects 'hope_test_0'  --net_dope vgg19-dcbb9e9d.pth --network 'dope'`
During training, the program issues an alarm
`WARNING:root:NaN or Inf found in input tensor.`
I tried lowering the lr and it didn't improve. 
In the end I used the trained network to identify my training set and unsurprisingly it didn't works correctly.
I uploaded my dataset samples, configuration files and obj files.
[test.zip](https://github.com/NVlabs/Deep_Object_Pose/files/8445056/test.zip)

mintar, 2022-04-08:
In your json files, the object has `"class": "test"`, so you need to specify `--objects test`.

I think you should also undo your change to `transforms.Resize`.

1920*1080 is much too large, it will be resized to something like 400x400 pixel anyway during training. Try 500x500 or 640x480 or something like that.

2000 images is not enough, try something like 10,000 or 20,000 or 60,000.

> During training, the program issues an alarm
> WARNING:root:NaN or Inf found in input tensor.

That's okay, this happens sometimes.

closed 2025-08-20 by nv-jeff